lorenzo-deluca/espkyogate #1: Bentel absoluta smart

Emilfoio, 2021-11-05:
Ciao Lorenzo, mi chiedevo se questo sistema per integrare l'allarme bentel su HA potesse andar bene sulla mia centrale smart absoluta. Qualcuno ne è a conoscenza? i passaggi da fare sono gli stessi? Grazie per il supporto.

dario81, 2021-11-05:
Ciao,
no non è compatibile con Absoluta.

lorenzo-deluca, 2021-11-05:
ciao, è una cosa che mi hanno chiesto in molti..
In realtà, come ho già detto, essendo l'Absoluta già in rete non serve nessun collegamento hardware, come questo che ho dovuto fare con hardware aggiuntivo.
Basta una integrazione software, mi pare che qui ne parlino [https://community.home-assistant.io/t/bentel-absoluta-alarm-system-integration/287646/48](url)

Io non ho un'Absoluta ma credo che sia relativamente semplice integrarla essendo tutto software.

commit af5d5f77 (2022-08-05): Try fix #1 and fid Documentation