lorenzo-deluca/espkyogate #2: Dispositivo non disponibile

manuele84, 2021-11-09:
Complimenti per il progetto lo aspettavo da tanto che qualcuno implementasse il componente. Ho eseguito tutta la procedura come indicato su un d1 mini1 pero in maniera random tutti le entità perdono la connessione per qualche secondo con home assistant e tutte le entità della centrale diventano non disponibili.
Ho riscontrato anche che il sensore mancanza rete e lo stato sirena sono invertiti.

In fase di compilazione riscontro questo warning.
[image]

lcavalli, 2021-11-09:
Anche a me è capitato durante lo sviluppo del codice. Ero al piano di sotto e l'access point configurato nel firmware era al piano di sopra. Il segnale era piuttosto debole e la connessione spesso cadeva, riagganciandosi poco dopo. Tieni conto anche della dimensione dell'antennina del modulo ESP8266, non proprio generosa. Io poi ho messo tutto in una scatoletta a sua volta dentro il box della centralina dell'allarme e penso che un po' attenuino il segnale.

I warning in compilazione sono, appunto, warning. Possono essere rimossi :smile:

lorenzo-deluca, 2021-11-09:
Ciao ragazzi,
i warning prima o poi li sistemerò... comunque non incidono sul funzionamento del firmware.

Credo anche io che il problema sia dovuto al WiFi, effettivamente non "tira" molto il wifi di quel modulo.
Direi che hai due possibilità, o cerchi di avvicinare l'access point alla centrale dell'allarme oppure cambi modulo e usi un modulo tipo il D1 mini Pro che ha un'antenna wifi esterna.

manuele84, 2021-11-10:
Grazie per i consigli, ho installato un repiter vicino la centrale e fin ora non ha dato più entità non disponibili.

Per caso avete riscontrato anche voi che  il sensore di mancanza rete è invertito con lo stato sirena?

lcavalli, 2021-11-10:
Nello yaml di esempio non mi pare:
```yaml
      return { 
        kyo32->kyo_comunication,
        kyo32->stato_sirena,
[...]
    binary_sensors:
      - id: kyo_comunication
        name: "Comunicazione Centrale"
      - id: stato_sirena
        name: "Stato Sirena"
```
Non li hai invertiti nella tua configurazione (l'ordine dei valori ritornati e l'ordine dei sensori definiti è importante)?

lorenzo-deluca, 2021-11-10:
Prova a verificare la configurazione come dice @lcavalli .
Se hai ancora problemi faccio un controllo, però ricordo che durante lo sviluppo avevo provato tutti i segnali e mi sembrava tutto ok. Fammi sapere 😉

Ropese, 2021-11-17:
Progetto stupendo a giorni sostituisco la norma 8 con una kyo e testo il tutto. L’unica cosa che non trovo e dove settare i nomi delle zone. Grazie

lorenzo-deluca, 2021-11-17:
ok, ottimo!
Intanto chiudo questa segnalazione ;)

commit e8251fac (2022-08-03): try fix #2
closed 2021-11-17 by lorenzo-deluca