lorenzo-deluca/espkyogate #3: Configuration trouble

brusarp, 2021-11-23:
Hello Lorenzo,
I am a recent and happy user of this supplement.
This isn't really an issue.
I'm trying to configure all areas and zones in the most correct and total way but something is not right ...
I noticed that if in the configuration I insert some areas or zones incorrectly in the home assistant "wrong" sensors are created (in a test yesterday, for example, "fuse fault" appeared to me as active but it was not true ...) it would not Is it possible to insert in the readme or in the espkyogate_configuraton.yaml file a configuration model where everything is commented? (currently for example there is only "Zone 1" configured) In this way people would have to do nothing but remove the comments they need to their taste ...
I don't know if I made myself clear ...

lorenzo-deluca, 2021-11-23:
Hello, it is a good tip.
I think your false "fuse fault" report may be due to the order of the sensors put in the configuration... 
Possibly you can post your configuration, only the binary_sensor section, of course...?

brusarp, 2021-11-23:
Sure, here you are:

`binary_sensor:
  - platform: custom
    lambda: |-
      auto kyo32 = new Bentel_Kyo32(id(uart_bus));
      App.register_component(kyo32);
      return { 
        kyo32->kyo_comunication,
        kyo32->stato_sirena,
        kyo32->warn_mancanza_rete,
        kyo32->sabotaggio_zona, 
        kyo32->sabotaggio_sistema,
        kyo32->zona_1,
        kyo32->zona_2,
        kyo32->zona_3,
        kyo32->zona_4,
        kyo32->zona_5,
        kyo32->zona_6,
        kyo32->zona_7,
        kyo32->zona_8,
        kyo32->zona_sabotaggio_1,
        kyo32->allarme_area_1,
        kyo32->allarme_area_2,
        kyo32->allarme_area_3,
        kyo32->allarme_area_4,
        kyo32->inserimento_totale_area_1,
        kyo32->inserimento_totale_area_2,
        kyo32->inserimento_totale_area_3,
        kyo32->inserimento_totale_area_4,
        kyo32->memoria_allarme_zona_1,
        kyo32->memoria_allarme_zona_2,
        kyo32->memoria_allarme_zona_3,
        kyo32->memoria_allarme_zona_4,
        kyo32->memoria_allarme_zona_5,
        kyo32->memoria_allarme_zona_6,
        kyo32->memoria_allarme_zona_7,
        kyo32->memoria_allarme_zona_8,
        kyo32->warn_scomparsa_bpi, kyo32->warn_fusibile, kyo32->warn_batteria_bassa, kyo32->warn_guasto_linea_telefonica, kyo32->warn_codici_default, kyo32->warn_wireless,
        kyo32->stato_uscita_1, kyo32->stato_uscita_2, kyo32->stato_uscita_3,
        kyo32->sabotaggio_bpi, kyo32->sabotaggio_jam, kyo32->sabotaggio_wireless};
    binary_sensors:
      - id: kyo_comunication
        name: "Comunicazione Centrale"
      - id: stato_sirena
        name: "Stato Sirena"
      - id: warn_mancanza_rete
        name: "Mancanza Rete"
      - id: sabotaggio_zona
        name: "Sabotaggio Zona"
      - id: sabotaggio_sistema
        name: "Sabotaggio Sistema"
      - id: zona_1
        name: "Zona 1"
      - id: zona_2
        name: "Zona 2"
      - id: zona_3
        name: "Zona 3"
      - id: zona_4
        name: "Zona 4"
      - id: zona_5
        name: "Zona 5"
      - id: zona_6
        name: "Zona 6"
      - id: zona_7
        name: "Zona 7"
      - id: zona_8
        name: "Zona 8"
      - id: zona_sabotaggio_1
        name: "Zona 1 Sabotaggio"
      - id: allarme_area_1
        name: "Allarme Area 1"
      - id: allarme_area_2
        name: "Allarme Area 2"
      - id: allarme_area_3
        name: "Allarme Area 3"
      - id: allarme_area_4
        name: "Allarme Area 4"
      - id: inserimento_totale_area_1
        name: "Inserimento Totale Area 1"
      - id: inserimento_totale_area_2
        name: "Inserimento Totale Area 2"
      - id: inserimento_totale_area_3  
        name: "Inserimento Totale Area 3"
      - id: inserimento_totale_area_4
        name: "Inserimento Totale Area 4"
      - id: memoria_allarme_zona_1
        name: "Memoria Allarme Zona 1"
      - id: memoria_sabotaggio_zona_1
        name: "Memoria Sabotaggio Zona 1"
      - id: warn_scomparsa_bpi
        name: "Scomparsa BPI"
      - id: warn_fusibile
        name: "Fusibile"
      - id: warn_batteria_bassa
        name: "Batteria Bassa"
      - id: warn_guasto_linea_telefonica
        name: "Guasto Linea Telefonica"
      - id: warn_codici_default
        name: "Codici Default"
      - id: warn_wireless
        name: "Wireless"
      - id: stato_uscita_1
        name: "Uscita 1"
      - id: stato_uscita_2
        name: "Uscita 2"
      - id: stato_uscita_3
        name: "Uscita 3"`

So if the problem is the order then I think it is all the more reason to "study" how to create a model where no line must be deleted otherwise it will skip everything ...

lorenzo-deluca, 2021-11-23:
Hi,
As I said, I'm not much of an expert on ESPHome.
In any case, from what I understand, the component configuration must be consistent between the return and the sensors being passed.
In your case it seems to me everything is ok until memoria_allarme_zona_1.
After that, as you can see, after that, as you can see, there is no congruence... right?
For example, memoria_sabotaggio_zona_1 it's not declared in return scope.

Can you check, fix and try again?

I agree with you that it's not exactly super simple, it can probably be done as you say but from the documentation on how to develop custom ESPHome components I couldn't find a smarter way.

brusarp, 2021-11-23:
Ok, checked and you are right (wrong declaration).
At this point I would say that to help you find a solution I will try to make a pull request with my idea...

lorenzo-deluca, 2021-11-23:
> Ok, checked and you are right (wrong declaration). At this point I would say that to help you find a solution I will try to make a pull request with my idea...

Great idea! Every pull request is welcome!

closed 2021-11-23 by lorenzo-deluca

brusarp, 2021-12-02:
Hello @lorenzo-deluca !
I sent you an email about this issue... I don't know if you've seen it... In short I can't get the Wemos D1 mini to work with all lines enabled...

commit 88bb4ec5 (2022-08-04): try fix #3