lorenzo-deluca/espkyogate #11: Errato messaggio di errore

amastrogiacomo1968, 2022-02-02:
Oggi sulla centralina è comparso il messaggio "Batteria Bassa" mentre in HA si è accesso il sensore "Guasto Linea Telefonica", cosa potrebbe essere successo? Grazie
[image]

lorenzo-deluca, 2022-02-12:
Ciao, sei sicuro di aver configurato correttamente i binary_sensor?
Tu hai la sempre Kyo32 giusto?
Eventualmente posta qui il tuo file di configurazione.. io non ho il combinatore telefonico ma è la prima volta che mi segnalano questa anomalia e controllando i documenti mi sembra che sia corretto il mapping.

amastrogiacomo1968, 2022-02-12:
espkyogate_configuration.yaml
`esphome:
  name: antifurto
  platform: ESP8266
  board: d1_mini
  includes:
    - bentel_kyo32.h

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.34
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32 Test Fallback Hotspot"
    password: "12345678"

captive_portal:

# Enable logging
logger:
  level: VERBOSE
  baud_rate: 0

# Enable Home Assistant API
api:

ota:

uart:
  id: uart_bus
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 9600
  data_bits: 8
  parity: EVEN

binary_sensor:
  - platform: custom
    lambda: |-
      auto kyo32 = new Bentel_Kyo32(id(uart_bus));
      App.register_component(kyo32);
      return { 
        kyo32->kyo_comunication, kyo32->stato_sirena, kyo32->warn_mancanza_rete, kyo32->sabotaggio_zona, kyo32->sabotaggio_sistema, kyo32->zona_1, kyo32->zona_2, kyo32->zona_3, kyo32->zona_4, kyo32->zona_5, kyo32->zona_6, kyo32->zona_7, kyo32->zona_8, kyo32->zona_9, kyo32->zona_10, kyo32->zona_11, kyo32->zona_12, kyo32->zona_13, kyo32->zona_14, kyo32->zona_15, kyo32->zona_16, kyo32->zona_17, kyo32->zona_18, kyo32->zona_19, kyo32->zona_20, kyo32->zona_sabotaggio_1, kyo32->allarme_area_1, kyo32->inserimento_totale_area_1, kyo32->inserimento_parziale_area_1, kyo32->memoria_allarme_zona_1, kyo32->memoria_sabotaggio_zona_1, kyo32->disinserita_area_1, kyo32->zona_sabotaggio_1, kyo32->allarme_area_2, kyo32->inserimento_totale_area_2, kyo32->inserimento_parziale_area_2, kyo32->memoria_allarme_zona_2, kyo32->memoria_sabotaggio_zona_2, kyo32->disinserita_area_2, kyo32->warn_scomparsa_bpi, kyo32->warn_fusibile, kyo32->warn_batteria_bassa, kyo32->warn_guasto_linea_telefonica, kyo32->warn_codici_default, kyo32->warn_wireless, kyo32->stato_uscita_1, kyo32->stato_uscita_2, kyo32->stato_uscita_3, kyo32->sabotaggio_bpi, kyo32->sabotaggio_jam, kyo32->sabotaggio_wireless};
    binary_sensors:
      - id: kyo_comunication
        name: "Comunicazione Centrale"
      - id: stato_sirena
        name: "Stato Sirena"
      - id: warn_mancanza_rete
        name: "Mancanza Rete"
      - id: sabotaggio_zona
        name: "Sabotaggio Zona"
      - id: sabotaggio_sistema
        name: "Sabotaggio Sistema"
      - id: zona_1
        name: "Porta Ingresso"
        device_class: door
      - id: zona_2
        name: "Riserva 1"
      - id: zona_3
        name: "Riserva 2"
      - id: zona_4
        name: "Riserva 3"
      - id: zona_5
        name: "Riserva 4"
      - id: zona_6
        name: "Riserva 5"
      - id: zona_7
        name: "Riserva 6"
      - id: zona_8
        name: "Riserva 7"
      - id: zona_9
        name: "Serranda TV"
        device_class: window
      - id: zona_10
        name: "Serranda Soggiorno"
        device_class: window
      - id: zona_11
        name: "Serranda Cucina Finestra"
        device_class: window
      - id: zona_12
        name: "Serranda Cucina Balconcino"
        device_class: window
      - id: zona_13
        name: "Serranda Bagno Rosso"
        device_class: window
      - id: zona_14
        name: "Serranda Studio"
        device_class: window
      - id: zona_15
        name: "Serranda Simone"
        device_class: window
      - id: zona_16
        name: "Serranda bagno Blu"
        device_class: window
      - id: zona_17
        name: "Serranda Camera da Letto"
        device_class: window
      - id: zona_18
        name: "Serranda Balcone"
        device_class: window
      - id: zona_19
        name: "Serranda Giorgio"
        device_class: window
      - id: zona_20
        name: "Riserva"
      - id: zona_sabotaggio_1
        name: "Zona 1 Sabotaggio"
      - id: allarme_area_1
        name: "Allarme Area 1"
      - id: inserimento_totale_area_1
        name: "Inserimento Totale Area 1"
      - id: inserimento_parziale_area_1
        name: "Inserimento Parziale Area 1"
      - id: memoria_allarme_zona_1
        name: "Memoria Allarme Zona 1"
      - id: memoria_sabotaggio_zona_1
        name: "Memoria Sabotaggio Zona 1"
      - id: disinserita_area_1
        name: "Disinserita Area 1"
      - id: allarme_area_2
        name: "Allarme Area 2"
      - id: inserimento_totale_area_2
        name: "Inserimento Totale Area 2"
      - id: inserimento_parziale_area_2
        name: "Inserimento Parziale Area 2"
      - id: memoria_allarme_zona_2
        name: "Memoria Allarme Zona 2"
      - id: memoria_sabotaggio_zona_2
        name: "Memoria Sabotaggio Zona 2"
      - id: disinserita_area_2
        name: "Disinserita Area 2"
      - id: warn_scomparsa_bpi
        name: "Scomparsa BPI"
      - id: warn_fusibile
        name: "Fusibile"
      - id: warn_batteria_bassa
        name: "Batteria Bassa"
      - id: warn_guasto_linea_telefonica
        name: "Guasto Linea Telefonica"
      - id: warn_codici_default
        name: "Codici Default"
      - id: warn_wireless
        name: "Wireless"
      - id: stato_uscita_1
        name: "Uscita 1"
      - id: stato_uscita_2
        name: "Uscita 2"
      - id: stato_uscita_3
        name: "Uscita 3"
`

amastrogiacomo1968, 2022-02-14:
Grazie @lorenzo-deluca , ho tolto le 2 precedenti risposte e ho caricato i 2 file di configurazione su pastebin per rendere la lettura più semplice.
Il file bentel_kyo32.h non è stato modificato [https://pastebin.com/6TrZVSVw](url)
Il file espkyogate_configuration.yaml è invece stato personalizzato con la mia configurazione [https://pastebin.com/GG7GvfBp](url).
Se ti dovesse servire, ma non credo, ti posto anche il file secrets.yaml .
Grazie come al solito del tuo aiuto

lorenzo-deluca, 2022-04-02:
Ciao @amastrogiacomo1968 , ho appena rilasciato una nuova versione del codice che oltre a compilare con le ultime versioni di ESPHome ho anche rivisto un po di cose.
Ho fatto una prova con la mia centrale da banco e la segnalazione di batteria bassa è corretta, non ho provato l'allarme del combinatore perché non ce l'ho ma ho verificato e dovrebbe essere ok anche quello.

Fammi sapere, se usi l'ultima versione del codice devi rivedere leggermente il tuo file yaml di configurazione ma è veramente poca cosa.

Grazie!

closed 2022-04-02 by lorenzo-deluca

amastrogiacomo1968, 2022-04-04:
Sto provando ad aggiornare la configurazione, ho una domanda.
Nel nuovo file espkyogate_configuration.yaml , alla riga 65, scrivi:
&kyo32->zona[0],

Io nel mio file (quello vecchio) ho devinite 20 zone in questo modo:
kyo32->zona_1, kyo32->zona_2, ........

Non capisco se devo ripetere la riga che metti tu n volte a definire le 20 zone che ho io o se invece l'esp le trova in automatico e le riconosce.

Grazie per il supporto

lorenzo-deluca, 2022-04-04:
Ciao,
no, devi specificarlo ed è cambiato, ora è zona[0], zona[1], etc.. trovi
l'esempio nel repository.

-- 

Regards,
*Lorenzo De Luca*

***@***.***
***@***.***>
*DevOps & **CTO*

dddprint.io

*PRIVACY INFORMATION*
The information contained in this e-mail message and/or in the enclosed
file/s is/are strictly
reserved. Their use is only allowed the addressee of the message, for the
purposes showed in the message itself. If you receive this message without
being its addressee, we kindly ask You to inform us about it by e-mail and
to proceed to the destruction of the same message cancelling it from Your
system.
It constitutes opposite behaviour to the principles prescribed by the
Legislative Decree 196/2003 the conservation of the message, the
divulgation and distribution of that message to other subjects, the copy of
it, or the use of it for different purposes.

P  Respects nature.
Do not print this email unless absolutely necessary.

lorenzo-deluca, 2022-04-04:
> Scusami non avevo visto l'esempio ma ero andato direttamente nel file
> .yaml :-)
>
> Però c'è una differnza tra esempio e quanto scritto nello
> espkyogate_configuration.yaml del repository
> In espkyogate_configuration.yaml scrivi:
> binary_sensors:
>
>    - id: zona_1
>    name: "Zona 1"
>
> Nell'esempio:
> binary_sensors:
>
>    - id: kyo_zone1
>    name: "Zone 1"
>
> Qual'è la sistassi corretta dell' id?
> Grazie

ciao, è un retaggio di copia-incolla.. in realtà l'id è a tua completa
discrezione.. puoi chiamarli come le vuoi.
La cosa IMPORTANTE è l'ordine, ovvero quello che metti nella return
(zona[0], zona[1], etc...) deve rispettare l'ordine con cui inserisci i
sensori.
Mentre il nome e l'id dei sensori non fa nessuna differenza.

In effetti non mi fa impazzire questo meccanismo perché può causare errori,
stavo valutando il cambio con un sistema in cui devi dichiarare ogni zona
tipo:
zone:
    - zona_1
    - zona_2

Non so quale sia la soluzione migliore, forse questa seconda, ma ancora
devo lavorarci per arrivare.

-- 

Regards,
*Lorenzo De Luca*

***@***.***
***@***.***>
*DevOps & **CTO*

dddprint.io

*PRIVACY INFORMATION*
The information contained in this e-mail message and/or in the enclosed
file/s is/are strictly
reserved. Their use is only allowed the addressee of the message, for the
purposes showed in the message itself. If you receive this message without
being its addressee, we kindly ask You to inform us about it by e-mail and
to proceed to the destruction of the same message cancelling it from Your
system.
It constitutes opposite behaviour to the principles prescribed by the
Legislative Decree 196/2003 the conservation of the message, the
divulgation and distribution of that message to other subjects, the copy of
it, or the use of it for different purposes.

P  Respects nature.
Do not print this email unless absolutely necessary.

amastrogiacomo1968, 2022-04-04:
Effettivamente qualche problema lo crea, devo fare dei controlli perché non funziona bene.
L'armo e il disarmo da HA funzionano bene ma quando armo e disarmo con la chiave elettronica non si aggiorna lo stato in HA, cosa posso controllare? Grazie