lorenzo-deluca/espkyogate #12: Problemi dopo update esphome

manuele84, 2022-02-16:
Ciao,
dopo l'ultimo aggiornamento del componente esphome 2022.02, il gateway ha cominciato ad avere problemi è diventato lento ed i sensori delle aree inserite non funzionano più.
Non visualizza nemmeno più nemmeno i log, restituendo un errore api.
In fase di compilazione restituisce una marea di warning. 

Ti allego il file estratto da esphome.
[logs_update-all.txt](https://github.com/lorenzo-deluca/espkyogate/files/8082194/logs_update-all.txt)

lorenzo-deluca, 2022-02-16:
Ciao, hai provato a usare l'ultima versione del mio codice [bentel_kyo32.h] che trovi qui sul repo?
Quale versione di ESPHome hai installato, 2022.2.0? Io con la 2022.1.4 non ho problemi.
Eventualmente, se non l'hai già fatto, usa l'ultima versione del mio codice, se non fosse risolutivo valuterei un downgrade di ESPHome.
Nei prossimi giorni proverò ad aggiornare ESPHome all'ultima versione che dovrebbe essere la 2022.2.0 e vedo se riesco a simulare e risolvere il problema.

Fammi sapere!

manuele84, 2022-02-16:
Si ho installato l'ultima versione di [bentel_kyo32.h] da qualche settimana e funzionava benissimo. 
L'aggiornamento che ho fatto oggi è 2022.2.0 ho provato a fare downgrade alla 2022.1.4 tramite un backup ma non torna alla precedente versione.

lcavalli, 2022-02-16:
Ciao, a me in passato era capitato di dover tornare a una versione precedente di esphome e avevo trovato docker molto comodo, basta fare il pull di un'immagine con tag specifico a una certa versione e lanciare il container da quella. Qui i dettagli: https://esphome.io/guides/faq.html#docker-reference

manuele84, 2022-02-16:
> Ciao, a passato era capitato di dover tornare a una versione precedente di e avevo trovato docker molto comodo, basta fare il pull di un'immagine con tag specifico in una certa versione e il contenitore da quella. Qui i dettagli: https://esphome.io/guides/faq.html#docker-reference

La mia installazione è su hassOs, posso lanciare ugualmente tramite terminale comandi docker?

lorenzo-deluca, 2022-02-17:
Si, puoi farlo.. alla fine gli addon non sono altro che dei Docker.. in ogni caso, se vuoi, ho visto che proprio oggi hanno rilasciato la 2022.2.1, magari risolve questo problema, puoi provare ad aggiornare?

Comunque i warnings che vedo nel log non sono niente di bloccante.. apparentemente non c'è motivo per cui non debba andare, magari è solo un problema di build,
In ogni caso io ti proporrei di aggiornare, se non è risolutivo cerco qualche indicazione su come fare il downgrade di un addon da HassOs.

Fammi sapere, grazie!

manuele84, 2022-02-17:
> Si, puoi farlo.. alla fine gli addon non sono altro che dei Docker.. in ogni caso, se vuoi, ho visto che proprio oggi hanno rilasciato la 2022.2.1, magari risolve questo problema, puoi provare ad aggiornare?
> 
> Comunque i warning che nel log non sono niente di bloccante.. non c'è motivo per cui vedo non dovrebbe andare, magari è solo un problema di build, magari è solo che è solo problema di build, magari è solo che sono niente di bloccante . su come fare il downgrade di un addon da HassOs.
> 
> Fammi sapere, grazie!

Grazie sei molto disponibile!
Ho provato ad aggiornare alla 2022.02.01 ma il problema è peggiorato l'esp risulta online su addon esphome invece su home assistan l'entità non sono disponibili.
Se provo a scaricare i log mi dà il seguente errore:
[image]

manuele84, 2022-02-17:
Sono riuscito a fare downgrade di esphome alla 2022.1.4 tramite docker ed è tornato tutto funzionante alla perfezione.
Se a qualcuno doveste servire la procedura tramite terminale su hassos è la seguente:
- docker images ---- per risalire alla versione esatta
- docker pull esphome/esphome-hassio-aarch64:2022.1.4 ----- scarica la vecchia versione
- docker rmi esphome/esphome-hassio-aarch64:2022.2.1 --- cancella la versione attuale
- docker tag esphome/esphome-hassio-aarch64:2022.1.4 esphome/esphome-hassio-aarch64:2022.2.1 ---- sostituisce la vervione

Grazie per i consigli rimango in attesa che aggiornate il progetto almeno procedo ad aggiornare esphome.

lorenzo-deluca, 2022-02-17:
Si scusa, i comandi te lì avrei detti io.. pensavo ci fosse modo direttamente da interfaccia grafica..
Bene comunque, almeno abbiamo appurato che il problema è nella nuova versione di ESPHome. Grazie!

Avevo contattato gli sviluppatori di ESPHome e mi avevano consigliato di utilizzare un altro approccio perché questo con tanti sensori potrebbe dare problemi.
A questo punto proverò direttamente l'altro approccio che richiede una riscrittura del codice e anche della configurazione, che però poi sarà più semplice dell'attuale..
Non garantisco sulle tempistiche ma magari per fine mese riesco a farlo.
Vi faccio sapere, lascio aperta la issue perché di fatto con la 2022.2.0+ non funziona..

lorenzo-deluca, 2022-02-17:
> @lorenzo-deluca io alla fine ho rifatto un componente analogo al tuo, ma la gestione dei sensori è diversa. Ho provato ad aggiornare all'ultima versione di esphome e a me funziona tutto. Appena ho un po' di tempo ripulisco la mia versione dai dettagli relativi al mio allarme e la rilascio così puoi confrontarla se ti serve.
Grande!
Se vuoi fai un fork del mio repo e fai una Pull Request che valuto la fattibilità... l'ho messo open source proprio perché sia un progetto di community ;)

lcavalli, 2022-02-17:
L'ispirazione è stata grazie a te (è scritto anche nel README!), però l'ho rifatto più che altro per imparare qualcosa di nuovo, quindi una PR forse è un po' invasiva. Devo ancora aggiungere il reset delle memorie, per ora non mi è ancora servito. Trovi il repository nella mia pagina github :)

lorenzo-deluca, 2022-02-18:
> L'ispirazione è stata grazie a te (è scritto anche nel README!), però l'ho rifatto più che altro per imparare qualcosa di nuovo, quindi una PR forse è un po' invasiva. Devo ancora aggiungere il reset delle memorie, per ora non mi è ancora servito. Trovi il repository nella mia pagina github :)

Ottimo! Bel codice, bravo! ;)
Hai già avuto modo di testarlo sulle altre centrali che ho visto gestisci? Kyo8, 8G etc..?
Io in realtà volevo fare un pesante refactor per gestire il tutto in una diversa modalità di integrazione, più nativa, su consiglio degli sviluppatori di ESPHome, perché mi hanno fatto notare altri utenti che se si prova a configurare tutto (vedi file Kyo_Full.yaml) l'ESP diventa instabile.. e gli stessi sviluppatori mi hanno detto che questa modalità risulta poco efficiente.
Tu per caso hai provato il tuo codice configurando tutto?

Come preferisci, io se vuoi sono disponibile a portare avanti un solo progetto, altrimenti ognuno per la sua strada 👍 
Se vuoi la mia mail è nel README ;)

manuele84, 2022-02-18:
> L'ispirazione è stata grazie a te (è scritto anche nel README!), però l'ho rifatto più che altro per imparare qualcosa di nuovo, quindi una PR forse è un po' invasiva. Devo ancora aggiungere il reset delle memorie, per ora non mi è ancora servito. Trovi il repository nella mia pagina github :)

Sembra un progetto interessante, sarebbe comodo inserire oltre arma in casa e fuori casa anche altre modalità in modo da personalizzare l'inserimento di varie aree. 

Appena ho un minuto provo ad installarlo su un nodemcu che mi avanza almeno faccio qualche prova 😅

lcavalli, 2022-02-18:
> Ottimo! Bel codice, bravo! ;) Hai già avuto modo di testarlo sulle altre centrali che ho visto gestisci? Kyo8, 8G etc..?

No, è solo una predisposizione. Alcuni pacchetti vanno interpretati in modo differente nel caso di altri modelli, ma non lo faccio. Dovrebbe essere semplice aggiungerle.

> Io in realtà volevo fare un pesante refactor per gestire il tutto in una diversa modalità di integrazione, più nativa, su consiglio degli sviluppatori di ESPHome, perché mi hanno fatto notare altri [...]

Io ho 11 zone. Però nel mio codice i sensori sono sempre creati tutti e 32. Non so se il fatto di esporne di meno verso HA ha un impatto in questo senso o meno.

> Come preferisci, io se vuoi sono disponibile a portare avanti un solo progetto, altrimenti ognuno per la sua strada 👍 Se vuoi la mia mail è nel README ;)

Io non ho moltissimo tempo in questo periodo, però possiamo unire le forze. Se vuoi puoi creare un repository inizialmente privato e proviamo assieme a integrare su quello e vediamo come va :)

brusarp, 2022-03-01:
Bravi ragazzi!! 👏🏻👏🏻👏🏻 Come @lorenzo-deluca sa già io sono disponibile come beta-tester (ho una Kyo32 a banco)... Inoltre, per cercare di ridurre le instabilità, avevo provato a passare a ESP32 (invece del ESP8266 di Wemos D1) ma senza successo... Tenetemi aggiornato su come procederete!!

amastrogiacomo1968, 2022-03-07:
Qualcuno ha provato ad aggiornare alle 2022.2.6? Il problema si presenta lo stesso? Grazie

lorenzo-deluca, 2022-03-23:
> Ciao, progetto molto interessante, purtroppo vedo che con esphome 2020. 2.2 ci sono problemi anche con la compilazione del firmware, sarei felice di contribuire come tester, ho molte centrali kyo 32 installate e potrei essere interessato al progetto. Resto in attesa, nel frattempo buon lavoro agli sviluppatori.

Ciao, hai provato con la versione come [descritto qui](https://github.com/lorenzo-deluca/espkyogate/issues/12#issuecomment-1043175237) ?
In ogni caso entro fine mese dovrei rimettere mano al codice e penso di sistemare questa incompatibilità.

Se vuoi parlare del progetto puoi contattarmi a me@lorenzodeluca.dev ;)

lorenzo-deluca, 2022-04-02:
Ciao a tutti, ho appena rilasciato una nuova versione che compila anche con le ultime versioni di ESPHome.
Già che c'ero (grazie anche a @lcavalli ) ho rivisto un po il codice e vi conviene riguardare l'esempio di yaml per la definizione delle zone (che ora vanno passate come array).
Per il resto dovrebbe essere tutto ok, fatemi sapere! 👍 

Grazie!

closed 2022-04-02 by lorenzo-deluca