lorenzo-deluca/espkyogate #13: Compatibilità con Kyo320G

alessandrocolos, 2022-03-15:
Ciao Lorenzo, progetto favoloso!
Seguo un tuo post su Facebook da parecchio tempo e ora mi sono deciso di acquistare tutto il materiale ma mentre osservavo il codice ho notato che è studiato per il modello 32gcon Max 32 zone; io possiedo una 320g, che supporta fino a 328 zone e 32 aree.
È possibile adattare il codice anche per questo modello?

lorenzo-deluca, 2022-04-02:
Ciao, riusciresti a provare questa ultima versione che ho rilasciato sulla tua centrale 320G?

lorenzo-deluca, 2023-02-13:
Ciao @alessandrocolos , hai per caso provato con la tua centrale la compatibilità?

lorenzo-deluca, 2023-09-29:
Ciao @alessandrocolos , sei poi riuscito a flashare ESPHome?
Ti consiglio di guardare qualche tutorial online prima su come flashare esphome e poi su come farlo utilizzando un custom_component come questo.

In realtà il procedimento è identico, solo che devi mettere nella directory del file yaml di configurazione Esphome anche il file components/bentel-kyo/bentel_kyo32.h che viene incluso nel progetto.

Però ti consiglio di studiarti un po e sperimentare ESPHome anche lato configurazione (yaml, etc..) perché poi se vuoi utilizzare per bene la mia integrazione ti conviene essere pratico altrimenti farai fatica a configurare le varie zone, aree, etc..

Fammi sapere ;)

closed 2023-09-29 by lorenzo-deluca