lorenzo-deluca/espkyogate #15: ESPHome v2022.3.1 - Impossible to compile

brembygit, 2022-03-26:
Today I tried to compile the code but is not possible, I also tried the latest _bentel_kyo32.h_ same error, any ideas?

Processing bentel-control (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0

In file included from src/main.cpp:195:0:
src/bentel_kyo32.h:241:70: error: no matching function for call to 'esphome::binary_sensor::BinarySensor::BinarySensor(const char [17])'
  BinarySensor *kyo_comunication = new BinarySensor("Kyo Comunication");
                                                                      ^
In file included from src/esphome/core/application.h:13:0,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor()
   explicit BinarySensor();
            ^
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note:   candidate expects 0 arguments, 1 provided
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor(const esphome::binary_sensor::BinarySensor&)
 class BinarySensor : public EntityBase {
       ^
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [17]' to 'const esphome::binary_sensor::BinarySensor&'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor(esphome::binary_sensor::BinarySensor&&)
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [17]' to 'esphome::binary_sensor::BinarySensor&&'
In file included from src/main.cpp:195:0:
src/bentel_kyo32.h:243:77: error: no matching function for call to 'esphome::binary_sensor::BinarySensor::BinarySensor(const char [22])'
  BinarySensor *warn_mancanza_rete = new BinarySensor("Warning Mancanza Rete");
                                                                             ^
In file included from src/esphome/core/application.h:13:0,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor()
   explicit BinarySensor();
            ^
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note:   candidate expects 0 arguments, 1 provided
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor(const esphome::binary_sensor::BinarySensor&)
 class BinarySensor : public EntityBase {
       ^
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [22]' to 'const esphome::binary_sensor::BinarySensor&'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor(esphome::binary_sensor::BinarySensor&&)
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [22]' to 'esphome::binary_sensor::BinarySensor&&'
In file included from src/main.cpp:195:0:
src/bentel_kyo32.h:244:77: error: no matching function for call to 'esphome::binary_sensor::BinarySensor::BinarySensor(const char [22])'
  BinarySensor *warn_scomparsa_bpi = new BinarySensor("Warning Scomparsa BPI");
                                                                             ^
In file included from src/esphome/core/application.h:13:0,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor()
   explicit BinarySensor();
            ^
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note:   candidate expects 0 arguments, 1 provided
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor(const esphome::binary_sensor::BinarySensor&)
 class BinarySensor : public EntityBase {
       ^
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [22]' to 'const esphome::binary_sensor::BinarySensor&'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor(esphome::binary_sensor::BinarySensor&&)
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [22]' to 'esphome::binary_sensor::BinarySensor&&'
[... 195 lines cut ...]
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [15]' to 'const esphome::binary_sensor::BinarySensor&'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor(esphome::binary_sensor::BinarySensor&&)
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [15]' to 'esphome::binary_sensor::BinarySensor&&'
In file included from src/main.cpp:195:0:
src/bentel_kyo32.h:257:76: error: no matching function for call to 'esphome::binary_sensor::BinarySensor::BinarySensor(const char [20])'
  BinarySensor *sabotaggio_wireless = new BinarySensor("Sabotaggio Wireless");
                                                                            ^
In file included from src/esphome/core/application.h:13:0,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor()
   explicit BinarySensor();
            ^
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note:   candidate expects 0 arguments, 1 provided
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor(const esphome::binary_sensor::BinarySensor&)
 class BinarySensor : public EntityBase {
       ^
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [20]' to 'const esphome::binary_sensor::BinarySensor&'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: esphome::binary_sensor::BinarySensor::BinarySensor(esphome::binary_sensor::BinarySensor&&)
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [20]' to 'esphome::binary_sensor::BinarySensor&&'
Compiling /data/bentel-control/.pioenvs/bentel-control/lib64d/WiFi/WiFiClient.cpp.o
Compiling /data/bentel-control/.pioenvs/bentel-control/lib64d/WiFi/WiFiGeneric.cpp.o
*** [/data/bentel-control/.pioenvs/bentel-control/src/main.cpp.o] Error 1
========================= [FAILED] Took 70.96 seconds =========================

lorenzo-deluca, 2022-03-30:
It is a known problem, I am working on a new version, for now you should compile it with a previous version of ESPHome https://github.com/lorenzo-deluca/espkyogate/issues/12#issuecomment-1043175237

lorenzo-deluca, 2022-04-02:
Ciao, ho appena rilasciato una nuova versione che dovrebbe funzionare con l'ultima versione di ESPHome.
Fammi sapere se ti funziona ;)

closed 2022-04-02 by lorenzo-deluca