lorenzo-deluca/espkyogate #16: Problema nuova release

manuele84, 2022-04-04:
Ciao ho provveduto ad installare la nuova configurazione, durante l'utilizzo riscontro i seguenti problemi:
-L'aggiornamento tramite OTA fallisce
-L'esp risulta instabile, si creano molte disconnessioni
-Lo stato delle partizioni dalla 3 in poi delle volte non viene sincronizzata, invece le prime due funzionano regolarmente.

Mi potresti dare qualche consiglio?

lorenzo-deluca, 2022-04-07:
Ciao, per stato delle partizioni intendi le zone?
Eventualmente puoi verificare se le prime due sulla centrale sono configurate in modo diverso dalle altre?
Tu avevi provato anche la precedente release e non avevi riscontrato questo problema sullo stato delle partizioni?

Per quanto riguarda l'instabilità anche io ho notato che non riesco a fare l'aggiornamento OTA ma devo collegarlo al PC.
Pensavo fosse il mio ESP, eventualmente faccio qualche altra verifica però a me non da problemi ne di instabilità ne di caduta di connessione, è acceso da 2 settimane in test senza mai una disconnessione.
Hai provato a riflashare l'ESP?

amastrogiacomo1968, 2022-04-07:
Confermo anche io l'impossibilità ad aggiornare OTA e le disconnessioni abbastanza frequenti con relativa indisponibilità vari sensori

manuele84, 2022-04-08:
> Ciao, per stato delle partizioni intendi le zone? Eventualmente puoi verificare se le prime due sulla centrale sono configurate in modo diverso dalle altre? Hai provato anche la versione precedente e non hai riscontrato questo problema sullo stato delle partizioni?
> 
> Per quanto riguarda l'instabilità anche io ho notato che non riesco a fare l'aggiornamento OTA ma devo collegarlo al PC. Pensavo fosse il mio ESP, eventualmente faccio qualche altra verifica però a me non da problemi ne di instabilità ne di caduta di connessione, è acceso da 2 settimane in test senza mai una disconnessione. Hai provato a rivedere l'ESP?

Ciao per le partizioni intendo l'area di inserimento, le prime due funzionano regolarmente la 3,4,5,6 quando le attivo cambiano stato, invece quando disinserisco l'allarme rimango sempre attive.
Con la precedente versione invece funzionavano, infatti per il momento sono tornato a quella.
Le zone invece ne ho configurate 16 e funzionavano tutte.
Sulla centrale le prime due sono configurate allo stesso modo delle altre

lorenzo-deluca, 2022-04-08:
Ok, grazie del feedback, se dici che sulla precedente funzionavano sicuramente è un errore dovuto alla migrazione, verificherò.
Quindi parliamo dei binary sensor da inserimento_totale_area_3 in poi?

lorenzo-deluca, 2022-04-08:
ciao ragazzi, ho rilasciato un aggiornamento che dovrebbe aver risolto l'instabilità.
Per quanto riguarda gli aggiornamenti OTA sembra una "novità" introdotta con le ultime versioni di ESPHome https://github.com/esphome/issues/issues/2629 in pratica la comunicazione seriale su TX/RX disturba gli aggiornamenti OTA.

Per ora ho visto che basta scollegare il pin del VCC che alimenta il modulo seriale e funziona.
In ogni caso cerco di sistemare intercettando gli aggiornamenti OTA e mettendo in pausa la comunicazione seriale con la centrale.
Un altro modo sarebbe quella di non usare TX/RX ma altri pin e configurare l'UART.

Riuscite a darmi un feedback in merito l'instabilità? Grazie!

amastrogiacomo1968, 2022-04-08:
Per le disconnessioni questo è il log di oggi
[Comunicazione Centralina Antifurto] acceso
13:32:31 - 2 ore fa
[Comunicazione Centralina Antifurto] è diventato non disponibile
13:32:28 - 2 ore fa
[Comunicazione Centralina Antifurto] acceso
11:34:52 - 4 ore fa
[Comunicazione Centralina Antifurto] è diventato non disponibile
11:34:43 - 4 ore fa
[Comunicazione Centralina Antifurto] acceso
10:55:53 - 5 ore fa
[Comunicazione Centralina Antifurto] è diventato non disponibile
10:55:50 - 5 ore fa
[Comunicazione Centralina Antifurto] acceso
10:52:23 - 5 ore fa

lorenzo-deluca, 2022-04-08:
Si, immaginavo.. puoi per caso provare ad aggiornare con l'ultimo .h che ho rilasciato? Grazie!

lorenzo-deluca, 2022-04-08:
Leggere i log della centralina è abbastanza macchinoso, tra l'altro credo sia abbastanza inutile perché se è collegata ad Home Assistant dovresti avere tutti i log nel registro home assistant decisamente più semplice da consultare, no?
Mentre per l'ora si, ci stavo lavorando ;)

lorenzo-deluca, 2022-04-08:
Per quanto riguarda l'aggiornamento OTA ho trovato questo workaround, senza dover scollegare nulla.
Nel file yaml dichiarare un input di safe mode:

`switch:
  - platform: safe_mode
    name: "ESPKyoGate (Safe Mode)"`

Quindi, quando si vuole fare l'upload andare nelle entità, attivare la safe mode e procedere con l'upload OTA.
Come vi sembra come procedura?
A me tutto sommato non sembra male, anche perché così si protegge l'esp da magari potenziali errori di upload.

Fatemi sapere! :)

manuele84, 2022-04-08:
> Ok, grazie del feedback, se dici che sulla precedente funzionavano sicuramente è un errore dovuto alla migrazione, verificherò. Quindi parliamo dei sensori binari da inserimento_totale_area_3 in poi?

Si esatto

lorenzo-deluca, 2022-04-08:
> > Ok, grazie del feedback, se dici che sulla precedente funzionavano sicuramente è un errore dovuto alla migrazione, verificherò. Quindi parliamo dei sensori binari da inserimento_totale_area_3 in poi?
> 
> Si esatto.. L'ultimo aggiornamento risolve il problema? Per lo switch della safe mode secondo me è un ottima soluzione 😁
Si, dovrebbe risolvere diversi problemi, puoi riprovare?

manuele84, 2022-04-08:
> > > Ok, grazie del feedback, se dici che sulla precedente funzionavano sicuramente è un errore dovuto alla migrazione, verificherò. Quindi parliamo dei sensori binari da inserimento_totale_area_3 in poi?
> > 
> > 
> > Si esatto.. L'ultimo aggiornamento risolve il problema? Per lo switch della modalità provvisoria secondo me è un'ottima soluzione😁
> 
> Si, si dovrebbe risolvere diversi problemi, si può riprovare?

Domani provo e ti faccio sapere...
Grazie

amastrogiacomo1968, 2022-04-08:
> Leggere i log della centralina è abbastanza macchinoso, tra l'altro credo sia abbastanza inutile perché se è collegata ad Home Assistant dovresti avere tutti i log nel registro home assistant decisamente più semplice da consultare, no? Mentre per l'ora si, ci stavo lavorando ;)

Si si nulla di macchinoso che poi finisce per funzionare male o per niente.
Per il resto mi sembra un ottimo lavoro, complimenti

amastrogiacomo1968, 2022-04-08:
> Per quanto riguarda l'aggiornamento OTA ho trovato questo workaround, senza dover scollegare nulla. Nel file yaml dichiarare un input di safe mode:
> 
> ```yaml
> switch:
>   - platform: safe_mode
>     name: "ESPKyoGate (Safe Mode)"
> ```
> 
> Quindi, quando si vuole fare l'upload andare nelle entità, attivare la safe mode e procedere con l'upload OTA. Come vi sembra come procedura? A me tutto sommato non sembra male, anche perché così si protegge l'esp da magari potenziali errori di upload.
> 
> Fatemi sapere! :)

Lo provo e ti faccio sapere, grazie

manuele84, 2022-04-09:
Ciao sta mattina ho fatto delle prove con l'ultima versione, il problema dello stato d' inserimento delle Aree rimane. Quando disattivi tutte l'aree  insieme, solo la 1 e la 2 comunicano il cambio di stato le restanti rimango nello stato inserito, potrebbe essere un problema di polling ? 
Invece lo swicht della safe mode funziona e  si riescono ad effettuare aggiornamenti OTA

brembygit, 2022-04-09:
Non so se sia lo stesso problema, ma non qualsiasi sensore tenti di aggiungere oltre a questi rende l'esp non funzionante
risponde al ping ma non si può accedervi (no console, no log) l'unica alternativa è ri-caricare via cavo il codice

```
binary_sensor:
  - platform: custom
    lambda: |-
      auto kyo32 = new Bentel_Kyo32(id(uart_bus));
      App.register_component(kyo32);
      return {

        kyo32->kyo_comunication,
        kyo32->stato_sirena,
        kyo32->warn_mancanza_rete,
        kyo32->warn_scomparsa_bpi,
        kyo32->warn_fusibile,
        kyo32->warn_batteria_bassa,
        kyo32->warn_guasto_linea_telefonica,
        kyo32->warn_codici_default,
        kyo32->warn_wireless,
        kyo32->sabotaggio_zona, 
        kyo32->sabotaggio_sistema,
        kyo32->sabotaggio_bpi,
        kyo32->sabotaggio_chiave_falsa,
        kyo32->sabotaggio_jam,
        kyo32->sabotaggio_wireless,
        
        &kyo32->zona[0],
        &kyo32->zona_sabotaggio[0],
        &kyo32->zona_esclusa[0],
        &kyo32->memoria_allarme_zona[0],
        &kyo32->memoria_sabotaggio_zona[0],
        
        &kyo32->allarme_area[0],
        &kyo32->inserimento_totale_area[0],
        &kyo32->inserimento_parziale_area[0],
        &kyo32->inserimento_parziale_ritardo_0_area[0],
        &kyo32->disinserita_area[0],
        
        &kyo32->stato_uscita[0]
        };

    binary_sensors:
      - id: kyo_comunication
        name: "Bentel Comunicazione Centrale"
      - id: stato_sirena
        name: "Bentel Stato Sirena"
      - id: warn_mancanza_rete
        name: "Bentel Mancanza Rete"
      - id: warn_scomparsa_bpi
        name: "Bentel Scomparsa BPI"
      - id: warn_fusibile
        name: "Bentel Fusibile"
      - id: warn_batteria_bassa
        name: "Bentel Batteria Bassa"
      - id: warn_guasto_linea_telefonica
        name: "Bentel Guasto Linea Telefonica"
      - id: warn_codici_default
        name: "Bentel Codici Default"
      - id: warn_wireless
        name: "Bentel Wireless"
      - id: sabotaggio_zona
        name: "Bentel Sabotaggio Zona"
      - id: sabotaggio_sistema
        name: "Bentel Sabotaggio Sistema"
      - id: sabotaggio_bpi
        name: "Bentel Sabotaggio BPI"
      - id: sabotaggio_chiave_falsa
        name: "Bentel Sabotaggio Chiave Falsa"
      - id: sabotaggio_jam
        name: "Bentel Sabotaggio Jam"
      - id: sabotaggio_wireless
        name: "Bentel Sabotaggio wireless"  
        
      - id: zona_1
        name: "Bentel Zona 1"

        
      - id: allarme_area_1
        name: "Bentel Allarme area 1"
      - id: allarme_area_2
        name: "Bentel Allarme area 2"
      - id: allarme_area_3
        name: "Bentel Allarme area 3"
      - id: allarme_area_4
        name: "Bentel Allarme area 4"
      - id: allarme_area_5
        name: "Bentel Allarme area 5"
      - id: allarme_area_6
        name: "Bentel Allarme area 6"
      - id: allarme_area_7
        name: "Bentel Allarme area 7"
        - id: allarme_area_8
        name: "Bentel Allarme area 8"

      - id: inserimento_totale_area_1
        name: "Bentel Inserimento Totale Area 1"      
      - id: inserimento_totale_area_2
        name: "Bentel Inserimento Totale Area 2"     
```

lcavalli, 2022-04-09:
Ciao, mi sembra che il numero di sensori ritornati dalla funzione lambda non torni con la successiva definizione (e significato).

brembygit, 2022-04-09:
> Ciao, mi sembra che il numero di sensori ritornati dalla funzione lambda non torni con la successiva definizione (e significato).

Non ho capito, puoi spiegarmi?

brembygit, 2022-04-09:
> Ciao, mi sembra che il numero di sensori ritornati dalla funzione lambda non torni con la successiva definizione (e significato).

Capito, a beneficio di tutti riporto la configurazione completa
Basta rimuovere le zone o le aree non usate
credo esita ancora il limite di 20 Zone, 8 Aree e 8 Uscite

```
binary_sensor:
  - platform: custom
    lambda: |-
      auto kyo32 = new Bentel_Kyo32(id(uart_bus));
      App.register_component(kyo32);
      return {

        kyo32->kyo_comunication,
        kyo32->stato_sirena,
        kyo32->warn_mancanza_rete,
        kyo32->warn_scomparsa_bpi,
        kyo32->warn_fusibile,
        kyo32->warn_batteria_bassa,
        kyo32->warn_guasto_linea_telefonica,
        kyo32->warn_codici_default,
        kyo32->warn_wireless,
        kyo32->sabotaggio_zona, 
        kyo32->sabotaggio_sistema,
        kyo32->sabotaggio_bpi,
        kyo32->sabotaggio_chiave_falsa,
        kyo32->sabotaggio_jam,
        kyo32->sabotaggio_wireless,
        
        &kyo32->zona[0],
        &kyo32->zona[1],        
        &kyo32->zona[2],        
        &kyo32->zona[3],        
        &kyo32->zona[4],        
        &kyo32->zona[5],        
        &kyo32->zona[6],                
        &kyo32->zona[7],
        &kyo32->zona[8],
        &kyo32->zona[9],        
        &kyo32->zona[10],        
        &kyo32->zona[11],        
        &kyo32->zona[12],        
        &kyo32->zona[13],        
        &kyo32->zona[14],                
        &kyo32->zona[15],
        &kyo32->zona[16],                
        &kyo32->zona[17],
        &kyo32->zona[18],                
        &kyo32->zona[19],    
        &kyo32->zona[20],
        &kyo32->zona[21],    
        &kyo32->zona[22],
        &kyo32->zona[23],    
        &kyo32->zona[24],
        &kyo32->zona[25],    
        &kyo32->zona[26],
        &kyo32->zona[27],    
        &kyo32->zona[28],
        &kyo32->zona[29],    
        &kyo32->zona[30],
        &kyo32->zona[31],    
        
        &kyo32->zona_sabotaggio[0],
        &kyo32->zona_sabotaggio[1],
        &kyo32->zona_sabotaggio[2],
        &kyo32->zona_sabotaggio[3],
        &kyo32->zona_sabotaggio[4],
        &kyo32->zona_sabotaggio[5],
        &kyo32->zona_sabotaggio[6],
[... 615 lines cut ...]
      - id: disinserita_area_6
        name: "Bentel Disinserita Area 6"
      - id: disinserita_area_7
        name: "Bentel Disinserita Area 7"
      - id: disinserita_area_8
        name: "Bentel Disinserita Area 8"   

      - id: stato_uscita_1
        name: "Bentel Stato Uscita 1"
      - id: stato_uscita_2
        name: "Bentel Stato Uscita 2"
      - id: stato_uscita_3
        name: "Bentel Stato Uscita 3"
      - id: stato_uscita_4
        name: "Bentel Stato Uscita 4"        
      - id: stato_uscita_5
        name: "Bentel Stato Uscita 5"
      - id: stato_uscita_6
        name: "Bentel Stato Uscita 6"
      - id: stato_uscita_7
        name: "Bentel Stato Uscita 7"
      - id: stato_uscita_8
        name: "Bentel Stato Uscita 8" 

```

lcavalli, 2022-04-10:
Non so se ci sia un limite imposto al numero di sensori oppure se dipenda dalle risorse della schedina esp8266. In ogni caso questi sono tutti binary sensor e possono essere facilmente aggregati (anzi, l'allarme li fornisce aggregati). Possono essere passati a HA sottoforma di un unico sensore (ad esempio tutti gli stati delle zone) e poi divisi in binary sensor usando un template tipo `{{ value | int | bitwise_and(1) }}` in HA.

lorenzo-deluca, 2022-04-10:
Esatto @brembygit , forse non mi sono spiegato bene io nella documentazione.
L'attuale implementazione prevede una "doppia" dichiarazione, sia nella return che nella lista e quello che fa fede non è il nome ma l'ordine.
Stavo lavorando su un'altra versione come integrazione nativa ma prima volevo essere sicuro che questa funzionasse correttamente.
Mi confemate quindi che non ci sono più instabilità?

Grazie mille @lcavalli , sempre sul pezzo! :)

amastrogiacomo1968, 2022-04-10:
> Mi confemate quindi che non ci sono più instabilità?

Purtroppo qualche disconnessione ancora avviene

Comunicazione Centralina Antifurto acceso
09:50:53 - 38 minuti fa
Comunicazione Centralina Antifurto è diventato non disponibile
09:50:49 - 38 minuti fa
Comunicazione Centralina Antifurto acceso
08:51:35 - 2 ore fa
Comunicazione Centralina Antifurto è diventato non disponibile
08:51:28 - 2 ore fa
Comunicazione Centralina Antifurto acceso
08:49:29 - 2 ore fa
Comunicazione Centralina Antifurto è diventato non disponibile
08:49:23 - 2 ore fa
9 aprile 2022
Comunicazione Centralina Antifurto acceso
21:51:53 - 13 ore fa
Comunicazione Centralina Antifurto è diventato non disponibile
21:51:47 - 13 ore fa
Comunicazione Centralina Antifurto acceso
16:02:56 - 18 ore fa
Comunicazione Centralina Antifurto è diventato non disponibile
16:02:54 - 18 ore fa
8 aprile 2022
Comunicazione Centralina Antifurto acceso
17:10:52 - L’altro ieri

lcavalli, 2022-04-10:
Riguardo all'instabilità della connessione, devo dire che io l'ho sempre vista. C'è una issue aperta nel repository di ESPHome proprio su questo argomento con un sacco di commenti e soluzioni di ogni tipo, ma niente di riolutivo. Secondo me però ha anche un forte impatto la dimensione dell'antenna WiFi della schedina. Se non ricordo male c'è la Wemos d1 mini pro che ha la possibilità di collegare un'antenna esterna. Puoi anche provare a mandare a HA un sensore con la potenza del segnale WiFi visto dall'ESP8266 come spiegato qui: https://esphome.io/components/sensor/wifi_signal.html?highlight=wifi_signal.

amastrogiacomo1968, 2022-04-11:
> Per quanto riguarda l'aggiornamento OTA ho trovato questo workaround, senza dover scollegare nulla. Nel file yaml dichiarare un input di safe mode:
>
> ```yaml
> switch:
>   - platform: safe_mode
>     name: [...]

Ho provato ad usare il "safe mode", ma come si usa?
L'ho attivato, ho aggiornato con successo ma poi tutto resta indisponibile come se restasse in safe mode.
Sbaglio io qualcosa?
Grazie

manuele84, 2022-04-12:
> Esatto @brembygit , forse non mi sono spiegato bene io nella documentazione. L'attuale implementazione prevede una dichiarazione "doppia", sia nella restituzione che nella lista e quello che fa fede non è il nome ma l'ordine. Stavo lavorando su un'altra versione come integrazione nativa ma prima volevo essere sicuro che questa funzionasse correttamente. Mi confemate quindi che non ci sono più instabilità?
> 
> Grazie mille @lcavalli , sempre sul pezzo! :)

Instabilità no.. Però non capisco come mai l'area dal due in poi non si sincronizza, con la vecchia versione quella funzionante con esphome 2022.1.4 andavano alla perfezione. Infatti per il momento sono tornato a quella.

lcavalli, 2022-04-12:
> Instabilità no.. Però non capisco come mai l'area dal due in poi non si sincronizza, con la vecchia versione quella funzionante con esphome 2022.1.4 andavano alla perfezione. Infatti per il momento sono tornato a quella.

Puoi provare ad abilitare i log di debug:

```yaml
# Enable logging
logger:
  level: DEBUG
  baud_rate: 0
```

Poi chiamare da HA il servizio:

```yaml
service: esphome.espkyogate_debug_command
data:
  serial_trace: 1
  log_trace: 1
```
Abilitando il log e il dump dei dati sulla seriale.
Oppure, se devi già aggiornare il firmware, imposta direttamente a true le variabili alla riga 262 del file .h

```c
bool serialTrace = true;
bool logTrace = true;
```

Infine abilita una delle zone incriminate, aspetta qualche secondo e disabilitala e salva il log.
Forse si riesce a capire se il problema è nel comando inviato alla centralina o nell'interpretazione dello stato.

manuele84, 2022-04-17:
> > Instabilità no.. Però non capisco come mai l'area dal due in poi non si sincronizza, con la vecchia versione quella funzionante con esphome 2022.1.4 andavano alla perfezione. Infatti per il momento sono tornato a quella.
> 
> Puoi provare ad abilitare i log di debug:
> 
> ```yaml
> # Abilita 
> logger di registrazione :
>    livello : DEBUG 
>   baud_rate : 0
> ```
> 
> Poi chiama da HA il servizio:
> 
> ```yaml
> servizio : esphome.espkyogate_debug_command 
> dati :
>    serial_trace : 1 
>   log_trace : 1
> ```
> 
> Abilitando il log e il dump dei dati sulla seriale. Oppure, se devi già aggiornare il firmware, imposta direttamente a true le variabili alla riga 262 del file .h
> 
> ```c
> bool serialTrace = true ;
> bool logTrace = true ;
> ```
> 
> Infine abilita una delle zone incriminate, aspetta qualche secondo e disabilitala e salva il log. Forse si riesce a capire se il problema è nel comando inviato alla centralina o nell'interpretazione dello stato.

Ho provato ad eseguire le prove consigliate sia su nodemcu v3 e su d1 mini ma in entrambe le schedine appena provo ad accedere hai log comincia un continuo rebot con relativa perdita di connessione. 
Forse c'è un limite di sensori max da esporre, il quale rende instabile l'esp? 
Attualmente ho esposto 50 sensori, con la vecchia versione funzionava bene.

lorenzo-deluca, 2022-05-29:
Ciao ragazzi,
mi pare che con l'ultima versione di ESPHome 2022.5 non ci siano più problemi con l'aggiornamento OTA.
Me lo potete confermare?

Grazie!

closed 2022-05-29 by lorenzo-deluca

lcavalli, 2022-06-03:
> Ciao ragazzi, mi pare che con l'ultima versione di ESPHome 2022.5 non ci siano più problemi con l'aggiornamento OTA. Me lo potete confermare?
> 
> Grazie!

Si, confermo. Ho provato con la 2022.5.1 e aggiorna OTA senza più bloccarsi.