lorenzo-deluca/espkyogate #17: Mancata individuazione di una zona

amastrogiacomo1968, 2022-04-07:
Ho aggiornato all'ultima versione e funziona tutto (anche il famoso sensore batteria scarica) tranne una zona che non riesco a fargli "trovare".
Posto qui https://pastebin.com/hKQbZ8jM il mio file e la zona che non riesco a fargli individuare è la 9 (Serranda Soggiorno), mi puoi aiutare a capire il motivo? Ho devinito 20 zone e funzionano tutte tranne questa.
Grazie

lorenzo-deluca, 2022-04-07:
Ciao, la configurazione mi sembra ok, per caso quella zona è configurata in modo diverso dalle altre sulla centrale?
Eventualmente appena ho modo provo a configuare anche io qualcosa sulla zona 9 (che poi dovrebbe essere il 10 dato che parte da zero).

amastrogiacomo1968, 2022-04-08:
Risolto, dopo avrer fatto l'ennesimo flash del Mini D1 (senza modificare il codice) la zona è comparsa. Chissà cosa è successo

closed 2022-04-08 by amastrogiacomo1968