lorenzo-deluca/espkyogate #21: kyo 8M

bogdanabaza, 2022-06-15:
Hi Lorenzo,
Thank you for your work on kyo integration on HA.
I have a kyo8M. I'm trying to connect it with esp32 and HA.
I made all the hardware setup. Now I have troubles on serial communication. I tried to use your bentel_kyo32.h but I it didn't wok.
The logs look like here:
[image]

Do you have any suggestion or hint please?
Tks,

lorenzo-deluca, 2022-06-15:
Hello ;)
there might be something to fix to handle the messages from Kyo 8.
Anyway, however, it seems to be unresponsive on the serial port.
Are the pins you have configured correct? have you tried reversing TX and RX yet?
The serial only works when you are not PROGRAMMING MODE.

Let me know! thanks ;)

bogdanabaza, 2022-06-15:
Yes, I reverse the TX and RX and no result. 
The programming mode is not activated.
Here is the pins:
[image]
The response is the same:
[image]

lorenzo-deluca, 2022-06-15:
So if it's okay on the serial side probably the Kyo 8 is not responding to that message, I tested this firmware only on Bentel Kyo32G..
If you feel like analyzing the protocol and you have Bentel software you can connect to it and see with software like serialmon what hexadecimal messages it accepts from Bentel software. it's not super easy but you can have fun ;)

closed 2022-06-15 by lorenzo-deluca

bogdanabaza, 2022-06-15:
Ok. 
I have Bentel software - Kyo Unit - and it works woth it. Before to look at the protocol in hexa I'll try to change the serial converter to be sure that is not from it.
Thank you!

lorenzo-deluca, 2022-12-14:
Hi @bogdanabaza, if you want to retry with the help of other users who have the Kyo8 unit, I worked on a version that should work.
For now, you can find it in this branch:  https://github.com/lorenzo-deluca/espkyogate/tree/hotfix/handle-kyo8

If you want to try it out and let me know how it goes.