lorenzo-deluca/espkyogate #22: Antisabotaggio non segnalato

amastrogiacomo1968, 2022-06-16:
Ciao Lorenzo e di nuovo complimenti per il tuo lavoro.
Ti volevo segnalare un "problema" con l'antisabotaggio.
Ieri stavo cambiando la batteria della sirena e sulla centralina è comparsa (correttamente) la segnalazione di "sabotaggio"
[image]
ma non si è attivato nessun binary_sensor a riguardo, ovviamente non è un problema bloccante ma mi sembra giusto segnalarlo.
Inoltre chiamando il servizio esphome.antifurto_reset_alarms l'allarme non si è resettato (ho fatto il reset da tastiera della centralina).
Se hai bisogno di altre info chiedi pure.
Grazie

lorenzo-deluca, 2022-06-20:
Ciao, strano, tu hai aggiunto tutti i sensori di sabotaggio e warnings ?

- warn_mancanza_rete
- warn_scomparsa_bpi
- warn_fusibile
- warn_batteria_bassa
- warn_guasto_linea_telefonica
- warn_codici_default
- warn_wireless
- sabotaggio_zona
- sabotaggio_chiave_falsa
- sabotaggio_bpi
- sabotaggio_sistema
- sabotaggio_jam
- sabotaggio_wireless

Perchè a parte questi non ne ho altri e sinceramente non so tutti i casi in cui esca quella X.
Tu eventualmente hai visto esattamente quale allarme è uscito?
Se hai staccato la batteria dovresti aver sicuramente visto il warn_batteria_bassa.

Fammi sapere

amastrogiacomo1968, 2022-06-21:
Buongiorno, i sensori sabotaggio e warn li ho tutti.
La X è comparsa quando ho aperto la sirena per sostituire la sua batteria (quindi si dovrebbero attivare dei sensori anti sabotaggio relativi alla sirena).
Il warn_batteria_bassa funziona e si è attivato quando si è scaricata la batteria della centralina e quando l'ho sostituita non si è più attivata (quindi funziona correttamente).
Spero di aver spiegato bene qual'è il problema (ovviamente non è bloccante :-))

lorenzo-deluca, 2024-04-26:
Ciao, ho testato con la mia centrale Kyo32G e l'antisabotaggio sistema viene correttamente segnalato con il binary_sensor.sabotaggio_sistema, puoi provare ad aggiornare all'ultima versione del mio firmware e farmi sapere?

Grazie! :)

closed 2024-04-26 by lorenzo-deluca