lorenzo-deluca/espkyogate #29: Sincronizzazione ora centralina con ora Home Assistastan

amastrogiacomo1968, 2022-12-22:
Non è una segnalazione di problema ma una richiesta di nuova funzionalità :-)
E' possibile sincronizzare l'ora della centralina Bentel con l'ora (sicuramente corretta) di Home Assistant?
Grazie

lorenzo-deluca, 2022-12-22:
Ciao, allora, in effetti avevo dimenticato di documentare il servizio per fare l'aggiornamento della data e ora.
Ora lo trovi in [documentazione ](https://github.com/lorenzo-deluca/espkyogate#update-unit-date-and-time)

Non prende automaticamente la data da home assistant ma se vuoi puoi fare tu un'automazione in cui prendi i parametri e lì valorizzi, anzi, se lo fai e mi dai lo script lo metto tra gli script in documentazione.

Una nota, mi pare non sia consigliato cambiare troppo spesso la data, forse non lo puoi neanche fare se l'allarme è inserito.

Eventualmente prova e fammi sapere come va, grazie! :)

closed 2022-12-22 by lorenzo-deluca

lcavalli, 2022-12-22:
Ciao @lorenzo-deluca, volevo chiederti se avevi qualche dettaglio in più sul fatto che sia sconsigliato aggiornare l'orario troppo spesso. Il mio componente lo fa in automatico col trigger on_time_sync della piattaforma homeassistant time. Hai idea di quali possano essere gli effetti?

lorenzo-deluca, 2022-12-22:
ciao @lcavalli , a dire il vero non ho documentazione in merito ma per esperienza si tende a non cambiare troppo frequentemente il clocktime perché magari la scheda ha delle logiche basate sul tempo (penso ad esempio a inserimenti/disinserimenti in base al giorno/ora) e potrebbe essere che queste vengano alterate.
Ora, non sapendo come è fatto il firmware della scheda magari non è influenzato ma non si sa mai, personalmente ho preferito lasciare solo il servizio a disposizione da chiamare alla bisogna.

amastrogiacomo1968, 2022-12-23:
> Ciao, allora, in effetti avevo dimenticato di documentare il servizio per fare l'aggiornamento della data e ora. Ora lo trovi in [documentazione [...]

provato e funziona, l'idea comunque è di utilizzarlo al cambio ora legale/solare o in caso di grosse differenza tra l'ora della centralina e l'ora reale. Grazie