lorenzo-deluca/espkyogate #30: Indisponibilità collegamento HA -> Cebtralina Bentel

amastrogiacomo1968, 2023-01-02:
Ciao Lorenzo, ti volevo segnalare un problema di "mancato collegamento" tra HA e la centralina Bentel.
Premetto che ho sempre notato queste temporane indisponiblità ma ultimaenti sono sempre più frequenti e lunghe; ulteriore premessa, ho una centralina Kyo 32, l'ultima versione del tuo sw e ho notato lo stesso problema con 2 MiniD1 (uno nuovissimo appena preso).
In pratica spesso manca il collegamento HA - Centralina con l'impossibilità quindi di leggere i sensori o inviare comandi, sto tracciando quanto problema e ti allego lo screen dove puoi vedere con la prima entità un sensore (inserimento totale area 1 ma varrebbe anche per gli altri) e il collegamento tra MiniD1 e router wifi (in pratica non si scollega mai dal wifi); come vedi ci sono spesso delle indisponibilità e una durata 9 ore circa e risolta solo riavviando HA.
Posso fare qualcosa per risolvere questo problema? Grazie
[image]

lorenzo-deluca, 2023-01-02:
Ciao, buon anno! 👍 
piuttosto strano, mi puoi dire esattamente versione di HA e di ESPHome?

Perché la modalità di agiornamento che utilizza ESPHome fa andare tutti i sensori  in "non disponibile" quando manca il collegamento tra ESP e HA, io ho un problema simile con la centrale in garage che è dovuta alla connessione tramite powerline, configurando un sensore ping configurato così: 
`count: 1
 scan_interval: 10`
 ho visto che effettivamente perdeva dei ping.
 
 Visto che ti capita così frequentemente prova a tenere aperti i log di ESPHome e vedere se anche lì viene tracciato qualcosa, se sono offline prova magari a utilizzare al posto di un MiniD1 un ESP32, qualche utente che aveva problemi di instabilità ha risolto così.

Fammi sapere

amastrogiacomo1968, 2023-01-02:
Buon anno anche a te :-)
Ho le ultime versioni di entrambi (Home Assistant 2022.12.7 e ESPHome 2022.12.3).
Per attivare i log basta tenere avviato ESPHome o devo fare altro?
Se volessi utilizzare un ESP32 devo fare una configurazione particolare o è identica a quella del MiniD1?
Grazie come al solito per la disponibilità

amastrogiacomo1968, 2023-01-02:
Aggiornamento forse risoluto :-)
Ho notato una saldatura sul MiniD1 fredda, ora lo rifatta e monitorizzo la situazione.
Altra info, nell'immagine che hai postato i collegamento noto che sono diversi da quelli che ho io (e mi sembra stano averne "sbagliati" 3 su 4); in pratica tu indichi di collegare:
- VCC su 3V3 mentro io l'ho collegato su 5V (filo rosso)
- TXD su D1 mentre io lo collego su TX (filo verde)
- RXD su D2 mentre io lo collego su RX (filo giallo)
Nonostate questo e al netto delle interruzioni (che potrebbero dipendere dalla saldatura fatta male) sembra funzionare tutto bene

lcavalli, 2023-01-02:
Per quanto riguarda l'alimentazione del convertitore TTL/RS-232 posso risponderti io. Il segnale VCC è collegato all'uscita 3.3V dell'ESP8266 perché il MAX3232 a differenza del MAX232 opera a 3.3V anziché a 5V. L'ideale sarebbe stato avere un oscilloscopio e poter misurare i segnali TX ed RX e verificare che fossero entro i 3.3V ammessi dai DIO dell'ESP8266 ed eventualmente adattarli. Durante i vari test io avevo erroneamente alimentato il MAX3232 con 5V e il chip era diventato caldissimo, per fortuna senza danni permanenti.

amastrogiacomo1968, 2023-01-02:
Grazie, quindi almeno per alimentazione da correggere. Mentre per le connessioni D1 e D2 mi sai dire qualcosa?

lcavalli, 2023-01-02:
Quelli o vanno oppure no. Quindi visto che la tua connessione va e viene sono per forza di cose giusti 🙂.

amastrogiacomo1968, 2023-01-03:
Infatti ;-)
Sbirciando nella foto pubblicata da Lorenzo [https://raw.githubusercontent.com/lorenzo-deluca/espkyogate/master/images/BentelKYO32G-Connections.jpg](url) direi che i collegamenti giusti sono quelli che ho fatto io e non quelli dello schema (questi per capirci [https://raw.githubusercontent.com/lorenzo-deluca/espkyogate/master/images/wiring.png](url) )

lorenzo-deluca, 2023-01-03:
Grazie a @lcavalli 👍 
In realtà io sul datasheet del modulo MAX3232 leggo tensione di ingresso accettata 3.3V-5.5V, onestamente io l'ho tenuto per anni alimentato a 5V (tant'è che nella foto è collegato così) e non ha mai dato problemi.
Onestamente avevo scelto 5V perché sul 3.3V mi pare che ogni tanto dava noia la seriale, ma magari era dovuta all'alimentazione non ottimale che avevo dato all'esp, a questo punto suppongo vadano bene entrambi i voltaggi.

Per quanto riguarda D1 e D2 era una variazione per non occupare i canali di TX e RX nel caso in cui ci si voglia collegare con una console seriale, come diceva anche Luca se funziona (anche se a "singhiozzo") la configurazione è corretta.

Diciamo che questi MiniD1 non è che siano proprio campioni di affidabilità, a me un paio hanno fatto fumo appena arrivati e collegati alla USB per il primo flash, probabilmente alla prima occasione lì sostituirò con dei ESP32 che invece mi sembrano più stabili.

Fammi sapere come va.

amastrogiacomo1968, 2023-01-03:
A questo punto provere ad usare un ESP32 che mi avanza.
Quello che ho io a solo la 3V3 quindi dovrò usare quella per alimentare il MAX232, per tx e rx utilizzo TXD0 e RXD0?

lorenzo-deluca, 2023-01-03:
Si, puoi usare i pin che preferisci, basta che descrivi i PIN nello yaml per la build di ESPHome.

amastrogiacomo1968, 2023-01-03:
Sto provando a programmare l'ESP32 dopo avere modificato il tuo lavoro (qui [https://pastebin.com/xP94tAD2](url)) ma quando provo installarlo sulla board ho questo errore che non capisco

ERROR Running command failed: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
ERROR Please try running esptool.py --before default_reset --after hard_reset --baud 460800 --port /dev/ttyACM0 --chip esp32 write_flash -z --flash_size detect 0x10000 /data/antifurto-esp32/.pioenvs/antifurto-esp32/firmware.bin 0x1000 /data/antifurto-esp32/.pioenvs/antifurto-esp32/bootloader.bin 0x8000 /data/antifurto-esp32/.pioenvs/antifurto-esp32/partitions.bin 0xe000 /data/cache/platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin locally.

lorenzo-deluca, 2023-01-03:
Mmmh, dice che non è avviato il modalità flash, di solito va premuto il tasto prima di connetterlo alla usb.

amastrogiacomo1968, 2023-01-03:
Ho provato a flasharlo dal pc e non dal server HA (un rasp) e sono riuscito, faccio qualche prova e ti aggiorno

amastrogiacomo1968, 2023-01-04:
Dopo quasi 24 ore di monitoraggio posso dire che la prova con l'ESP32 al posto del MiniD1 ha risolto i problemi di micro indiponibilità che avevo
[image]

Per chi fosse interessato ad utilizzare un ESP32 al posto del MiniD1 è sufficiente sostiruire al codice proposto da @lorenzo-deluca  le seguenti righe:

`esphome:
  platform: esp32
  board: esp32dev`

closed 2023-01-04 by amastrogiacomo1968