lorenzo-deluca/espkyogate #31: Comunicazione LAN

Eddie979, 2023-03-01:
Ciao Lorenzo, complimenti per il progetto.
Ho una Kyo32 e stavo da tempo cercando come renderla compatibile con home assist.
Volevo chiederti se esiste una versione di processore ESP compatibile con il tuo progetto che possa di poter connettersi anche via LAN oltre che Wi-Fi (ESP32?)

lorenzo-deluca, 2023-03-02:
Ciao @Eddie979 , grazie ;)
Avevo usato delle shield ethernet, tipo quel W5500 di cui parli tu però dovresti fare un mini circuito perché probabilmente avrai VCC e GND in comune con il modulo RS232, ma credo funzioni.
Altrimenti se vuoi puoi provare questo modulo [WT32-ETH01]( https://www.amazon.it/Wireless-Tag-WT32-ETH01-Development-Networking-Bluetooth/dp/B09BVJ72BT) che non ho mai usato ma sembra che possa fare al caso tuo, è un all-in-one che mi sembra più comodo poi da installare dentro la centrale.

Fammi sapere come va ed eventualmente se ti va fai qualche foto così la aggiungo alla documentazione :)

Eddie979, 2023-03-02:
Grazie mille Lorenzo😊Appena installo documento e ti invio delle foto👍🏻Ora sto attrezzando la rete con un mini pc persone assist e lavorare con ESP home.Grazie ancora- -Cordiali saluti/Best regardsEdgar Verga(Sent from my iPhone)

Eddie979, 2023-03-02:
…solo una domanda suli collegamenti:Nello schema sul tuo repo ho visto che la UART non è collegata a TX ed RX bensì a GPIO D1 e D2; nell’altra foto reale con la scheda kyo 32G è invece ben visibile il collegamento a TX ed RX, è coretto lo schema?:Grazie anticipatamente - -Cordiali saluti/Best regardsEdgar Verga(Sent from my iPhone)

lorenzo-deluca, 2023-03-02:
Si hai ragione, le immagini non sono allineate.. in realtà è indifferente, dipende dalla scheda che usi.
Quello che fa fede è la configurazione di ESPHome:
`uart:
  id: uart_bus
  tx_pin: GPIO5
  rx_pin: GPIO4`

Eddie979, 2023-03-02:
Grazie per la tua disponibilità e competenza!L’ultima domanda:WT32-ETH01 sembra avere un processore ESP32 anziché ESP8266; in questo caso il tuo codice funzionerebbe comunque? Ovvero, sono entrambi compatibili i processori con il tuo progetto?Grazie anticipatamente - -Cordiali saluti/Best regardsEdgar Verga(Sent from my iPhone)

lorenzo-deluca, 2023-03-02:
Figurati, allora, in realtà questo progetto è un'estensione di ESPHome, quindi è compatibile con tutti i dispositivi compatibili con ESPHome, [qui trovi una lista ](https://esphome.io/devices/index.html), è compatibile anche con [ESP32 ](https://esphome.io/devices/nodemcu_esp32.html), non parla specificamente di quel device ma suppongo che con la corretta configurazione degli IO funzioni, ovviamente da testare ;)

Eddie979, 2023-03-09:
Ciao Lorenzo,Finalmente ci siamo, o quasi.Ho installato tutto, la compilazione va a buon fine dopo qualche problemino: avendo usato un WT32 LAN ho dovuto modificare qualcosa nel file di configurazione .yaml e nel codice in C++ non veniva riconosciuto l’array “byte” e ho dovuto sostituire con uint8, il significato non cambia ma al compilatore non piaceva, credo per via di una differente versione del compilatore per il WT32Mi resta ancora un’ultima errore da correggere, ti chiedo cortesemente se mi puoi dare una mano: alla fine della compilazione compare l’errore “ERROR Connecting to espkyogatebox.local:3232 failed:  [Errno 111] Connection refusedDi seguito un’immagine:Grazie anticipatamente, attendo un tuo suggerimento.- -Cordiali saluti/Best regardsEdgar Verga(Sent from my iPhone)

Eddie979, 2023-03-09:
…ci sono riuscito!😊😊😊Ho trovato sul forum HA altri che come me svevanonanslo problema; semplicemente per la prima volta WT32 vuole essere programmato dalla macchina e non OTA; risolto, grazie.- -Cordiali saluti/Best regardsEdgar Verga(Sent from my iPhone)

Eddie979, 2023-03-09:
…devo però aggiornare il fw della kyo a 2.13, il mio è 2.10😪Ho letto sul tuo troubleshooting che potrebbe essere causa di malfunzionamento - -Cordiali saluti/Best regardsEdgar Verga(Sent from my iPhone)

lorenzo-deluca, 2023-03-09:
Ciao @Eddie979, l'aggiornamento del firmware della centrale potrebbe non essere necessario.
Sei riuscito a collegarla? Vedi dei dati su Home Assistant?
Ti consiglio per cominciare a configurare solo una zona e provare se vedi il cambio di stato su HA.

Eddie979, 2023-03-09:
Ciao Lorenzo,si, sono riuscito a comunicare con la centrale e vedo il cambio di stato ad esempio del sabotaggio centrale.Sto provando ora ad aggiungere zone per vedere se cambiano gli stati. (Se li aggiungo nel file .yaml durante la compilazione mi genera errore; probabilmente sbaglio qualcosa).Il mio impianto è composto da 1kyo32G con 8 switch collegati (finestre e porte); poi ha 3 espansioni Bentel collegate sulla stessa linea di comunicazione usata per le tastiere; ad ognuna di queste espansioni sono collegati altri switch e sensori movimento PIR.Per riassumere, in totale il mio impianto è suddiviso in 5 aree ognuna delle quali ha configurato switch e sensori PIR; quindi andando ad armare una specifica area includo PIR e switch configurati per quell’area.Il tuo progetto prevede anche la gestione delle espansioni Bentel collegate a Kyo32G?- -Cordiali saluti/Best regardsEdgar Verga(Sent from my iPhone)

closed 2023-05-21 by lorenzo-deluca