lorenzo-deluca/espkyogate #32: espkyogate KYO32 (non G)

popeye70, 2023-05-21:
Ciao Lorenzo,
innanzitutto complimentoni per il tuo bel lavoro!

Ho appena collegato la mia centrale d'allarme in oggetto funziona, se vuoi aggiungerla alla lista di compatibilità.
Con lo yaml di esempio, solo leggermente adattato al mio hardware e wifi, HA mi ha fatto attivare una integrazione che ha trovato n.1 dispositivi e n.28 entità ripartite tra Sensori, Registro e Configurazione.
Purtroppo "lato Bentel" non ho accesso alla configurazione del mio allarme, so solo che è composto da 32 sensori porte/finestre/persiane suddivisi in 3 gruppi attivabili/disattivabili separatamente;
presumo quindi che ora devo implementare lo yaml aggiungendo dei blocchi identici a "id: zona_1" ma numerati da 2 fino a 32, e poi testare pazientemente aperture e chiusure in giro per casa per verificare chi corrisponde a che cosa...è corretto?
Grazie

Flavio

lorenzo-deluca, 2023-05-21:
Ciao Flavio, grazie!
In che senso non hai accesso alla configurazione? Non hai il pin per entrare in programmazione?
Per configurare lo YAML ti basta sapere la posizione delle zone, comunque come dici ovviamente vanno da 1 a 32, puoi anche configurarle tutte e vedere empiricamente (provando a violare le varie zone) la posizione e associargli un nome.

Fammi sapere come va.

lorenzo-deluca, 2023-05-22:
Che scheda stai usando?
Mi pare che in molti abbiano risolto con una ESP32 al posto della Wemos D1.. DHCP o IP fisso non fanno differenza.
Solo considera che l'antenna WiFi di questi dipositivi è piuttosto scarsa.

popeye70, 2023-05-22:
Sto usando un D1, e la centralina del KYO32 è a circa 1,5 metri da un reperater wifi mesh collegato via powerline al router Fritzbox 7590.
Non ho sottomano degli ESP32 quindi per ora procedo col D1, e poi vedremo. Provando ad aggiungere la zona_2 mi si inchioda la scheda con questo errore:

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from esphome-web-d3fc12.local using esphome API
WARNING Can't connect to ESPHome API for esphome-web-d3fc12.local: Error resolving IP address: [Errno -5] No address associated with hostname
INFO Trying to connect to esphome-web-d3fc12.local in the background

Cancellando la nuova zona è ricaricando lo yaml, ha ripreso a funzionare correttamente. Probabilmente sbaglio o dimentico qualcosa nella sintassi, ho compilato semplicemente così:

      - id: zona_1
        name: "Persiane bagno 1P"
      - id: zona_1_sabotaggio
        name: "Sabotaggio Persiane bagno 1P"
      - id: zona_1_esclusa
        name: "Escluse Persiane bagno 1P"
      - id: zona_1_memoria_allarme
        name: "Memoria Allarme Persiane bagno 1P"
      - id: zona_1_memoria_sabotaggio
        name: "Memoria Sabotaggio Persiane bagno 1P"
      - id: zona_2
        name: "Finestra camera 1P"
      - id: zona_2_sabotaggio
        name: "Sabotaggio Finestra camera 1P"
      - id: zona_2_esclusa
        name: "Esclusa Finestra camera 1P"
      - id: zona_2_memoria_allarme
        name: "Memoria Allarme Finestra camera 1P"
      - id: zona_2_memoria_sabotaggio
        name: "Memoria Sabotaggio Finestra camera 1P"

Cosa mi sto perdendo secondo te?
grazie.

lorenzo-deluca, 2023-05-23:
Dai, buon lavoro! :)

closed 2023-05-23 by lorenzo-deluca

popeye70, 2023-12-24:
Ciao Lorenzo,
dopo alcuni mesi di funzionamento regolare devo rimettere mano alla configurazione perchè causa lavori in casa il mio allarme che prima era composto da 5 aree e 32 sensori ora è composto da sole 3 aree e 26 sensori.
Pensavo di cavarmela cancellando semplicemente le righe non più necessarie, ma dopo avere ricaricato la configurazione ottengo l'errore:

[15:52:27][W][component:214]: Component <unknown> took a long time for an operation (0.11 s).
[15:52:27][W][component:215]: Components should block for at most 20-30ms.
[15:52:27][W][component:214]: Component <unknown> took a long time for an operation (0.11 s).
[15:52:27][W][component:215]: Components should block for at most 20-30ms.

Dallo screenshot del software Bentel che mi ha lasciato l'elettricista si evince che non ha riprogrammato tutto l'antifurto da capo, ma ha solo cancellato quanto non più necessario lasciando dei "buchi" nelle zone:

[image]

e anche le aree 2 e 4 non sono state cancellate, ma semplicemente nessuna delle zone attualmente presenti è associata ad esse. Questi buchi possono creare qualche problema?

Nel programmare l'esp per soli 26 sensori/zone devo numerare gli array:
1) da &kyo32->zona[0] a &kyo32->zona[25]
2) da &kyo32->zona[1] a &kyo32->zona[31]  saltando i numeri mancanti
3) da &kyo32->zona[0] a &kyo32->zona[31] com'era prima quando avevo 32 sensori?

e poi di conseguenza come mi comporto nella sezione binary_sensors per gli id, che prima andavano da zona_1 a zona32?

Mille grazie per l'aiuto, e tanti auguri di Buone Feste

popeye70, 2023-12-29:
Ottimo grazie mille, ora è tornato tutto a posto!

Per quanto riguarda le aree e le zone attive potrei quindi anche cancellare tutti i "&kyo32->zona_esclusa[n]" e tutti i "kyo32->inserimento_parziale_area[n]" e "&kyo32->inserimento_parziale_ritardo_0_area[n]? Così eviterei la creazione di un bel pò di sensori che per il mio utilizzo del Bentel sono superflui...

lorenzo-deluca, 2023-12-29:
> Ottimo grazie mille, ora è tornato tutto a posto!
>
> Per quanto riguarda le aree e le zone attive potrei quindi anche cancellare tutti i "&kyo32->zona_esclusa[n]" e tutti i [...]

Si esatto, puoi configurare puntualmente solo i sensori che ti interessa vedere su Home Assistant

ruimarinho, 2024-04-12:
@popeye70 since the Bentel website is down for a while, would you be able to share the "KYO Unit" software with me please? I need to check the firmware version of my board. Cheers.

popeye70, 2024-04-12:
Hi Rui,

it's still online here: http://www.soloallarmi.it/istruzioni/bentel/BentelSecuritySuite-Installer-5.5.4007.exe

popeye70, 2024-04-12:
> Thank you @popeye70! Did you have to update your firmware on your unit?

No, i don't. Why you need to upgrade the firm? Espkyogate is not working for you?

ruimarinho, 2024-04-12:
I have a KYO32M but reading materials online in Italian appears to indicate that my unit is a KYO32 with a "M"etal casing. The README mentions that the KYO32G needs at least firmware 2.13 so I'm going to try and use the software you posted to read the board information directly. Hopefully that will get my espkyogate to work..