lorenzo-deluca/espkyogate #33: Integrazione homeassistant

dadelinux, 2023-06-08:
Ciao Lorenzo,
cercando in rete un modo per interfacciarmi alla mia centrale Kyo32 ho trovato il tuo progetto e ho subito acquistato i componenti hardware per costruire l'interfaccia.
Non ho capito come fare ad integrare questo repository in homeassistant, quali sono i passi per metterlo in funzione?
Grazie
Davide T.

brusarp, 2023-06-09:
> Ciao Lorenzo, cercando in rete un modo per interfacciarmi alla mia centrale Kyo32 ho trovato il tuo progetto e ho subito acquistato i componenti hardware per costruire l'interfaccia. Non ho capito come fare ad integrare questo repository in homeassistant, quali sono i passi per metterlo in funzione? Se ho intuito bene per flashare il modulo ESP devo prima copiare la cartella bentel-kyo con il suo contenuto e il file espkyogate_configuration.yaml all'interno della cartella config/esphome su homeassistant?
> 
> Grazie Davide T.

Quasi tutto corretto,
le cartelle dovranno avere questa forma:

config\esphome\bentel-kyo

mentre il file espkyogate_configuration.yaml è una bozza della configurazione da correggere (secondo le proprie necessità come wifi, sensori binari, configurazione uart ecc...)  e poi da utilizzare nell'interfaccia di EspHome.

dadelinux, 2023-06-13:
Ciao Lorenzo,
sono riuscito ad interfacciarmi e in homeassistant vedo gli stati di vari sensori. Non ho però capito questa parte di configurazione: ( dome sono configurate queste entità?)

- type: entities
    entities:
      - entity: binary_sensor.porta
        secondary_info: last-updated
      - entity: binary_sensor.radar_living
        secondary_info: last-updated
      - entity: binary_sensor.radar_camera
        secondary_info: last-updated
      - entity: binary_sensor.radar_mansarda
        secondary_info: last-updated
      - entity: binary_sensor.radar_lavanderia
        secondary_info: last-updated
      - entity: binary_sensor.persiana_bagno
        secondary_info: last-updated
      - entity: binary_sensor.persiana_cucina
        secondary_info: last-updated
      - entity: binary_sensor.persiana_living
        secondary_info: last-updated
    state_color: true
    show_header_toggle: false

Naturalmente non li vedo nel mio HomeAssistant ma per logica pensavo di ritrovarmi i sensori magnetici presenti sulle finestre...  Mi sai aiutare su come impostare in HA i sensori collegati sulla centrale?

[image]

lorenzo-deluca, 2023-06-13:
Ciao @dadelinux ,
non mi è chiarissimo il problema, se dici di riuscire a vedere gli stati dei vari sensori  dovresti vederli anche in questa scheda.. non è che il dispositivo ESPKyoGate poi per qualche motivo va non disponibile?

Se così fosse prova a usare un ESP32 al posto del WemosD1

dadelinux, 2023-06-13:
Forse non mi sono spiegato bene... ad esempio il binary_sensor.porta o un'altro sopra elencato dove lo hai dichiarato?
in homeassistant facendo una ricerca non lo trovo. Se guardo nel .yaml non c'è quindi da qualche altra parte lo avrai dichiarato

brusarp, 2023-06-14:
> Ciao Lorenzo, sono riuscito ad interfacciarmi e in homeassistant vedo gli stati di vari sensori. Non ho però capito questa parte di configurazione: ( dome sono configurate queste entità?)
> 

Ricorda che ogni entità va "dichiarata" nel file 2 volte (@lorenzo-deluca correggimi...) :

-  1 volta nell'elenco lambda (una riga per entità preceduta da &kyo32->)

- 1 volta nell'elenco binary sensor nello stesso ordine e per comodità usando come id lo stesso dichiarato nel lambda (più che altro per un motivo mnemonico).
Tieni conto che la kyo prevede la definizione della prima zona come "0" che poi mnemonicamente nei sensor diventa "1"...

Quindi se sono 2 zone:
       &kyo32->zona[0],
        &kyo32->zona_sabotaggio[0],
        &kyo32->zona_esclusa[0],
        &kyo32->memoria_allarme_zona[0],
        &kyo32->memoria_sabotaggio_zona[0],
        &kyo32->zona[1],
        &kyo32->zona_sabotaggio[1],
        &kyo32->zona_esclusa[1],
        &kyo32->memoria_allarme_zona[1],
        &kyo32->memoria_sabotaggio_zona[1],

  binary_sensors:
      - id: zona_1
        name: "Garage"
        device_class: "garage_door"
      - id: zona_1_sabotaggio
        name: "Sabotaggio Garage"
      - id: zona_1_esclusa
        name: "Esclusione Garage"
      - id: zona_1_memoria_allarme
        name: "Memoria Allarme Garage"
      - id: zona_1_memoria_sabotaggio
        name: "Memoria Sabotaggio Garage"
      - id: zona_2
        name: "Radar Taverna"
        device_class: "motion"
      - id: zona_2_sabotaggio
        name: "Sabotaggio Radar Taverna"
      - id: zona_2_esclusa
        name: "Esclusa Radar Taverna"
      - id: zona_2_memoria_allarme
        name: "Memoria Allarme Radar Taverna"
      - id: zona_2_memoria_sabotaggio
        name: "Memoria Sabotaggio Radar Taverna"

dadelinux, 2023-06-14:
Grazie mille per i chiarimenti, ora è tutto chiaro! questa sera proverò a riconfigurare per bene il file yaml dopo essermi interfacciato con la Kyo32G tramite Security Suite per vedere Aree/Zone.
Domanda: dopo aver modificato il file yaml posso eseguire l'aggiornamento tramite OTA? L'operazione va a buon fine?

lorenzo-deluca, 2023-06-14:
Ciao, bene, per quanto riguarda l'aggiornamento purtroppo ho notato che da qualche versione di ESPHome a questa parte la procedura OTA ogni tanto si incastra, questo perché l'interfaccia seriale è occupata a parlare con la centrale.
Di solito funziona però se da noia prima di far partire l'upload puoi attivare la Safe Mode, dovrebbe averti creato un pulsante all'interno del dispositivo EspKyoGate

[image]

Facci sapere come va ;)

brusarp, 2023-06-14:
Io non ho mai avuto problemi di aggiornamenti OTA (considerate che utilizzo un ESP32s però... gestisco 105 entità...)

[image]

lorenzo-deluca, 2023-06-14:
Si, penso che il problema sia dovuto al Wemos D1 che è particolarmente lento a gestire seriale e wifi.

dadelinux, 2023-06-15:
Ieri mi sono interfacciato alla Kyo32G con Security Suite e ho visualizzato le zone, vi riporto uno screen sotto. Da quello che si vede  è presente un expander alloggiato in una scatola a muro dove sono collegati dei sensori ( nr. 15-23 ) invece di essere direttamente sui morsetti della centrale. Non essendo numericamente continui ( 1-2-3-4.-.....-15-16-17-18-19-20-21-22-23 ) devo creare un file yaml con tutti i 32 sensori? ( esempio utilizzando il config Kyo_full.yaml:  [https://github.com/brembygit/espkyogate/blob/master/Kyo_Full.yaml](url) presente nel fork dibrembygit ) oppure posso saltare le posizioni vuote? 
[image]

lorenzo-deluca, 2023-06-15:
Ciao, assolutamente puoi saltare le posizioni vuote ;)
Ho fatto apposta con l'array posizionale per non dover mappare sempre tutto :)
Non guardare l'esempio di dibrembygit perché usa il vecchio mapping, ora invece come diceva @brusarp si usa l'array che parte da 0 (che sarebbe la zona 1).

Fammi sapere come va ;)

dadelinux, 2023-06-16:
Ciao ragazzi, Progettino FANTASTICO!
Ho finalmente interfacciato la mia Kyo32G in modo corretto con aree e tutti i sensori di zona come da indicazioni di @lorenzo-deluca e @brusarp
Ho abbozzato il mio Lovelance Panel nel modo più funzionale possibile :)
[image]

lorenzo-deluca, 2023-06-17:
@dadelinux  Ottimo, bel pannello!
Se vuoi puoi collaborare al progetto aggiungendo feature/documentazione, ovviamente mettere una stella al repo⭐ e se vuoi contribuire [qui](https://www.buymeacoffee.com/lorenzodeluca)

Alla prossima! ;)

closed 2023-06-17 by lorenzo-deluca