lorenzo-deluca/espkyogate #34: Reset Memoria Allarme Zone

dadelinux, 2023-06-21:
@lorenzo-deluca  dopo un po' di test ho notato che il comando di reset memoria allarme zone non funziona. Ho analizzato i log e la stringa seriale viene mandata regolarmente:
21:51:53][I][reset_alarms:154]: Reset Alarms.
[21:51:53][I][sendMessageToKyo:649]: TX '0F.05.F0.01.00.05.07.00.07 (9)'
[21:51:53][I][sendMessageToKyo:658]: RX '0F.05.F0.01.00.05 (6)'

La centrale risponde:
[21:51:53][E][reset_alarms:158]: kyo respond 6

Da home assistant eseguo il service:
service: esphome.kyo32gateway_reset_alarms
data: {}

Questo mi crea un problema xchè solo da tastierino posso resettare le memorie e non posso farlo da remoto.
Qualche idea?

lorenzo-deluca, 2023-06-22:
Ciao @dadelinux, avevo testato il comando di reset memoria allarmi e funzionava.
Tu hai Kyo32G? Mi puoi dire che versione firmware hai sulla centrale?

dadelinux, 2023-06-22:
Ciao @lorenzo-deluca , la centrale è una Kyo32G e monta l'ultima versione del firmware: 2.13
Posso fare qualcosa per darti ulteriori dati?

lorenzo-deluca, 2023-06-22:
Ok, dovrebbe essere come le mie.
Nei prossimi giorni faccio un test e verifico i comandi..

lorenzo-deluca, 2023-06-23:
Ciao @dadelinux , ho fatto una rapida analisi del protocollo seriale, potrebbe essere che il messaggio di reset memoria allarmi abbia hardcoded di applicarlo solo sulle prime 3 aree.
Per caso la memoria allarme che provi a resettare è di un'area che non è nelle prime 3?
Per controprova, puoi provare con la tua centrale a far scattare un allarme in Area 1 e a fare reset tramite HA?
Mi aspetto che funzioni, fammi sapere ;)

dadelinux, 2023-06-23:
@lorenzo-deluca, è come dici tu, le memorie allarme delle zone appartenente alle area 1,2,3 vengono resettate mentre quelle dell'area 4 no. E' possibile abilitare il reset anche sulle memorie dell'area 4?

closed 2023-06-24 by lorenzo-deluca

lorenzo-deluca, 2023-06-24:
@dadelinux , dovrei averla fixata.. se aggiorni il firmware con l'ultima versione v1.4.1
Magari dammi un feedback perché nelle mie centrali non ho più di 3 aree e non mi sono messo a crearne di nuove per fare il test ma mi aspetto funzioni.

Aspetto tue 👍

brusarp, 2023-06-24:
Confermo il funzionamento nel mio impianto (Kyo32G) con area 4 in allarme.
Ottimo lavoro come sempre @lorenzo-deluca 👍👍👍

dadelinux, 2023-06-24:
Grande @lorenzo-deluca ! Confermo anch'io! reset memorie allarme sulle zone di tutte le 4 aree funzionante!

PR #36 (2023-06-24): fix reset memory alarm for all areas
Fix #34