lorenzo-deluca/espkyogate #35: Alarm Control Panel

dario81, 2023-06-22:
Ciao Lorenzo,
ho visto che nella 2023.6.0 di EspHome hanno rilasciato l'Alarm Control Panel e Template Alarm Control Panel. Sarebbe davvero bello integrarlo nel progetto.

lorenzo-deluca, 2023-06-22:
Ciao @dario81, non sapevo di questa novità di ESPHome.
Ho dato un occhio veloce alla documentazione, sicuramente è interessante però faccio fatica a trovarne il valore aggiunto.
Mi pare che sia utile per far diventare l'ESP l'equivalente di una micro centrale di allarme, visto che la logica alla fine è piuttosto semplice.
Ma nel momento in cui ti colleghi ad una centrale già funzionante, come nel caso di questa integrazione con Bentel Kyo, quale sarebbe il vantaggio?
Devi comunque definire tutte le zone, aree, etc..

Potrebbe essere utile creare un package.yaml o altro per far si che una volta configurato ESPHome su Bentel uno si possa creare il pannello e automazioni con notifiche, etc.. io ho messo in documentazione un pannello di esempio ma è sicuramente migliorabile.

Tu che dici?

Grazie

lcavalli, 2023-06-22:
Ciao, forse ti riferisci alla nuova grafica della dialog box dell'alarm control panel di HA 2023.4?

[image]

Per quella basta definire un template alarm control panel come descritto qui https://www.home-assistant.io/integrations/alarm_control_panel.template dove associ alle varie azioni i servizi che l'integrazione mette a disposizione e allo stato dell'allarme un pezzo di codice nel template che riporti le varie stringhe `disarmed`, `armed_home`, `armed_away`, `arming`, etc.. Forse per lo stato serve qualche modifica all'integrazione, perchè non mi pare che questa informazione venga esposta verso HA. Questo farà si che l'integrazione venga vista da HA come un sistema di allarme e non solo come una serie di sensori e quindi possa venir usata con i componenti grafici specifici.

dario81, 2023-06-23:
Ciao @lorenzo-deluca,
più o meno fa la stessa cosa del template di HA, ma lo fa direttamente a livello di firmware e integrazione di ESPHome e lo espone direttamente a HA come se fosse un allarme. Un po' quello che dici tu nella seconda parte del messaggio sopra. Se trovo il tempo ci guardo in maniera più approfondita.

lorenzo-deluca, 2023-06-23:
Si, il valore aggiunto sta nel fatto che viene creato un pannello di allarme ma se non ho visto male tutto il resto: sensori, aree, zone, etc.. ovviamente devi definirli manualmente.
Come giustamente dice @lcavalli è l'equivalente di creare un pannello di allarme e collegargli le entità che l'integrazione ESPHome ti mette a disposizione.
Io ad esempio l'avevo fatto quando ancora HA non era così evoluto sfruttando l'integrazione Alarmo, che sto ancora usando perché mi sembra ancora più completa rispetto a quanto HA metta a disposizione nativamente.

Quindi preferirei lasciare più libertà possibile agli utenti, a discapito forse di essere leggermente meno plug & play. 
Voi che ne pensate?

Poi, come già detto sopra, sicuramente sarebbe di aiuto per gli utenti meno esperti preparare un package yaml con già automazioni e pannelli preconfigurati, lascio la palla a voi ;)

closed 2023-06-23 by lorenzo-deluca

lcavalli, 2023-06-28:
@dario81 scusa, ho realizzato adesso che ti riferivi a ESPHome, non Home Assistant!