lorenzo-deluca/espkyogate #37: KYO32M Support

ruimarinho, 2023-07-30:
Hi,

Thank you for such an interesting project @lorenzo-deluca! I have a **KYO32M** and I was wondering if you think it'll be compatible with this module?

I've ordered the components and will definitely test it on my own, but wanted to get early feedback if possible. Or maybe someone else in the community tried that model already?

ruimarinho, 2023-08-24:
@lorenzo-deluca I've went ahead and tried espkyogate on this unit. So far, motion sensor information seems to be flowing, everything else no (e.g. if any area is armed or disarmed). I'm going to log serial and share with you results. In the meantime, while everyone is sleeping (no movement), this is what serial is outputting:

```
[00:18:37][I][debug_command:167]: serial_trace 1 log_trace 1 polling_kyo 1
[00:18:37][I][sendMessageToKyo:649]: TX 'F0.02.15.12.00.19 (6)'
[00:18:37][I][sendMessageToKyo:658]: RX 'F0.02.15.12.00.19.FF.00.08.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.07 (26)'
[00:18:37][I][sendMessageToKyo:649]: TX 'F0.04.F0.0A.00.EE (6)'
[00:18:37][I][sendMessageToKyo:658]: RX 'F0.04.F0.0A.00.EE.00.00.00.00.00.00.00.00.00.00.00.00 (18)'
[00:18:38][I][sendMessageToKyo:649]: TX 'F0.02.15.12.00.19 (6)'
[00:18:38][I][sendMessageToKyo:658]: RX 'F0.02.15.12.00.19.FF.00.08.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.07 (26)'
```

How does it look to you?

ruimarinho, 2023-09-29:
@lorenzo-deluca any chance you could help me with this one? Eager to get it working :)

lorenzo-deluca, 2023-09-29:
Hi @ruimarinho , sorry, I had missed your message.
I do not know the differences of central "M" versus "W" or "G" models.
Matter is the central reply correctly to the status request message but everything seems to be zero.

The only way would be to connect with Bentel software from PC "Kyo Unit" to the central and through the central monitoring function figure out which status request.
Otherwise, if you want to use integration online you can find second-hand a lot of already tested Kyo32/Kyo32G centrals (electronic boards only) that work perfectly with integration ;) 

Let me know

closed 2023-09-29 by lorenzo-deluca

ruimarinho, 2023-11-20:
@lorenzo-deluca to be sure, this is not a wiring problem? The differences between "M" and "G" series seems to be minimal, which shouldn't explain this behaviour. Anything else I can attempt before looking into changing hardware?

ruimarinho, 2024-04-11:
@lorenzo-deluca do you have access to the Kyo Unit software that I could use to debug this? I have had zero success in finding second hand board so far..