lorenzo-deluca/espkyogate #41: Allarme batteria bassa?

popeye70, 2023-09-28:
Ciao Lorenzo,
Sto usando da qualche mese questa tua integrazione con la mia KYO32 (non G, se vuoi aggiungerla alla compatibility list) e funziona come un gioiellino.
Mi chiedevo se fosse possibile, oltre agli allarmi e sabotaggi già presenti, implementare anche l'allarme batteria bassa dei sensori wireless, che sarebbe una gran comodità per individuare al volo quale sia la batteria da sostituire...
Complimenti ancora per quanto hai realizzato.

Flavio

lorenzo-deluca, 2023-09-29:
Ciao @popeye70 , ottimo, aggiungo alla lista ;)
Tu per caso sai se questa informazione si vede dal software di gestione Bentel?
Perché via seriale riesco a leggere solo quello che è visibile al software Bentel... se mi dici come fare per leggerlo da lì in qualche modo riesco a gestirlo.

Grazie! :)

popeye70, 2023-09-29:
È un messaggio che appare sul display del tastierino, quindi credo del tutto simile agli allarmi e sabotaggi che hai già intercettato...proprio ieri mi è apparso "batteria bassa Z14" ed è lì che mi sono accorto che HA non lo replicava.

lorenzo-deluca, 2023-09-29:
Si, su HA vedi eventualmente i warnings ma sono generici.
Everntualmente hai modo di verificare ora sul software Bentel?

popeye70, 2023-09-29:
Per leggere la KYO col software Bentel è necessario collegarle direttamente il pc tramite un convertitore USB-seriale immagino...niente wifi eh?
No al momento non riesco sorry, ma vedrò cosa posso fare...

lorenzo-deluca, 2023-09-30:
Ahah, Wifi...!
Si esatto, devi collegarti con cavo seriale.. io ora non ho sensori wireless con batteria bassa.
Dovesse succedermi proverò a collegarmi via seriale e vedere se per caso c'è nelle segnalazioni ma non ricordo di averlo mai visto, secondo me è solo un log di evento nel registro eventi, quello da leggere via seriale in real-time è troppo grosso..

closed 2023-09-30 by lorenzo-deluca

popeye70, 2023-12-29:
> _Ciao @popeye70 , ottimo, aggiungo alla lista ;) Tu per caso sai se questa informazione si vede dal software di gestione Bentel? Perché via seriale riesco a leggere solo quello che è visibile al software Bentel... se mi dici come fare per leggerlo da lì in qualche modo riesco a gestirlo._

Ciao Lorenzo,
aggiorno questo thread per confermarti che la batteria bassa sensori nel software Bentel si vede, è la quarta colonna di questo jpg, e ti confermo anche che quando inserisci una batteria nuova nel sensore il colore di quel semaforo si aggiorna istantaneamente:

[image]

Le prime tre colonne sono gli stati di Allarme, Sabotaggio ed Esclusa che già sei riuscito a leggere dalla seriale...pensi quindi sia possibile recuperare anche questa info? Sarebbe fantastico!