lorenzo-deluca/espkyogate #42: Kyo 320

TheJim0, 2023-10-15:
Buondì,questa integrazione può funzionare anche con la kyo320?

TheJim0, 2023-10-15:
Ho provato a caricare il file yaml solo con gli eventi di sistema senza le zone e mi appare questo messaggio nei log
[image]

lorenzo-deluca, 2023-10-15:
Ciao, è da testare ;)
Allora, dall'errore sembra che non risponda proprio nulla, hai già provato a invertire TX e RX?
Non sei in programmazione della centrale giusto?

Poi, per un log della porta seriale più dettagliato dovresti abilitare il logging seriale:

```yaml
service: esphome.espkyogate_debug_command
data:
  serial_trace: 1
  log_trace: 1
  polling_kyo: 1
```

TheJim0, 2023-10-15:
Premetto che sono un installatore e programmatore anche di Bentel, ho provato anche ad invertirli, sono su tx e ex del d1 mini come GPIO1 e 3 rispettivamente

TheJim0, 2023-10-15:
Mi viene da pensare se tra la kyo32 e la 320 ci siano delle configurazioni diverse magari

lorenzo-deluca, 2023-10-15:
È possibile ma la prima chiamata seriale penso sia sempre la stessa perché è quella che chiede le informazioni sulla centrale.
La configurazione seriale quindi è ok? GPIO 1 e 3 sono configurati anche nella parte seriale?

Eventualmente prova ad abilitare i log seriali e fai uno screen.

TheJim0, 2023-10-15:
In che senso nella parte seriale?
Ma i log lo metto nella configurazione yaml?

TheJim0, 2023-10-15:
Ok ho fatto, il log però rimane identico

TheJim0, 2023-10-15:
In che senso nella parte seriale?
Ho chiamato il servizio e ora esce così

lorenzo-deluca, 2023-10-15:
Sicuro di aver chiamato il servizio giusto? 
Penso che sia diverso il servizio, sarà service: esphome.espkyogatebox_debug_command

Dopo averlo chiamato nei log dovresti vedere TX e DX

TheJim0, 2023-10-15:
Sisi ora è partito così[image]

lorenzo-deluca, 2023-10-15:
Niente, sembra non rispondere.. se sei sicuro sui pin di TX e RX mi sa che non gli piace il messaggio di stato. Se sei pratico dovresti provare a collegarti con il PC e il software Bentel e sniffare quello che passa sulla seriale ma non è proprio banale.

TheJim0, 2023-10-15:
Provo a vedere se riesco a fare qualcosa, nel caso ti aggiorno 👍

lorenzo-deluca, 2024-04-26:
Chiudo per inattività.

closed 2024-04-26 by lorenzo-deluca