lorenzo-deluca/espkyogate #43: Errore durante la compilazione

fabix683, 2023-10-29:
Salve Lorenzo.
Sto provando a compilare il tuo progetto tramite ESPHOME

Durante la compilazione ricevo i seguenti messaggi d'errore.
Grazie

`INFO ESPHome 2023.10.3
INFO Reading configuration /config/esphome/esphome-web-53c388.yaml...
Failed config

binary_sensor.custom: [source /config/esphome/esphome-web-53c388.yaml:51]
  platform: custom
  lambda: !lambda |-
    auto kyo32 = new Bentel_Kyo32(id(uart_bus));
    App.register_component(kyo32);
    return {
      kyo32->kyo_comunication,
      kyo32->stato_sirena,
      kyo32->warn_mancanza_rete,
      kyo32->sabotaggio_zona, 
      kyo32->sabotaggio_sistema,
      kyo32->warn_scomparsa_bpi,
      kyo32->warn_fusibile,
      kyo32->warn_batteria_bassa,
      kyo32->warn_guasto_linea_telefonica,
      kyo32->warn_codici_default,
      kyo32->warn_wireless,
      kyo32->sabotaggio_bpi,
      kyo32->sabotaggio_jam,
      kyo32->sabotaggio_wireless,
      kyo32->sabotaggio_chiave_falsa,
    
      &kyo32->zona[0],
      &kyo32->zona[1],
      &kyo32->zona[2],
      &kyo32->zona[3],
    
      &kyo32->zona_sabotaggio[0],
      &kyo32->zona_sabotaggio[1],
      &kyo32->zona_sabotaggio[2],
      &kyo32->zona_sabotaggio[3],
    
      &kyo32->zona_esclusa[0],
      &kyo32->zona_esclusa[1],
      &kyo32->zona_esclusa[2],
      &kyo32->zona_esclusa[3],
    
      &kyo32->memoria_allarme_zona[0],
      &kyo32->memoria_allarme_zona[1],
      &kyo32->memoria_allarme_zona[2],
      &kyo32->memoria_allarme_zona[3],
    
      &kyo32->memoria_sabotaggio_zona[0],
      &kyo32->memoria_sabotaggio_zona[1],
      &kyo32->memoria_sabotaggio_zona[2],
      &kyo32->memoria_sabotaggio_zona[3],
    
      &kyo32->allarme_area[0],
      &kyo32->allarme_area[1],
      &kyo32->allarme_area[2],
      &kyo32->allarme_area[3],
    
      &kyo32->inserimento_totale_area[0],
      &kyo32->inserimento_totale_area[1],
      &kyo32->inserimento_totale_area[2],
      &kyo32->inserimento_totale_area[3],
    
      &kyo32->inserimento_parziale_area[0],
      &kyo32->inserimento_parziale_area[1],
      &kyo32->inserimento_parziale_area[2],
      &kyo32->inserimento_parziale_area[3],
    
      &kyo32->inserimento_parziale_ritardo_0_area[0],
      &kyo32->inserimento_parziale_ritardo_0_area[1],
      &kyo32->inserimento_parziale_ritardo_0_area[2],
      &kyo32->inserimento_parziale_ritardo_0_area[3],.
    
      &kyo32->disinserita_area[0],
      &kyo32->disinserita_area[1],
      &kyo32->disinserita_area[2],
      &kyo32->disinserita_area[3],
    
      &kyo32->stato_uscita[0]
      &kyo32->stato_uscita[1]
      &kyo32->stato_uscita[2]
      &kyo32->stato_uscita[3]
    };
  binary_sensors: 
    - id: kyo_comunication
      name: Comunicazione Centrale
      disabled_by_default: False
    - id: stato_sirena
      name: Stato Sirena
      disabled_by_default: False
    - id: warn_mancanza_rete
      name: Mancanza Rete
      disabled_by_default: False
    - id: sabotaggio_zona
      name: Sabotaggio Zona
      disabled_by_default: False
    - id: sabotaggio_sistema
      name: Sabotaggio Sistema
      disabled_by_default: False
    - id: warn_scomparsa_bpi
      name: Scomparsa BPI
      disabled_by_default: False
    - id: warn_fusibile
      name: Fusibile
      disabled_by_default: False
    - id: warn_batteria_bassa
      name: Batteria Bassa
      disabled_by_default: False
    - id: warn_guasto_linea_telefonica
      name: Guasto Linea Telefonica
      disabled_by_default: False
    - id: warn_codici_default
      name: Codici Default
      disabled_by_default: False
    - id: warn_wireless
      name: Wireless
      disabled_by_default: False
    - id: sabotaggio_bpi
      name: Sabotaggio BPI
      disabled_by_default: False
    - id: sabotaggio_jam
      name: Sabotaggio Jam
      disabled_by_default: False
    - id: sabotaggio_wireless
      name: Sabotaggio wireless
      disabled_by_default: False
    - id: sabotaggio_chiave_falsa
      name: Sabotaggio Chiave Falsa
      disabled_by_default: False
    - id: zona_1
      name: Sensore DT Taverna
      device_class: motion
      disabled_by_default: False
    - id: zone_2
      name: Finestra Taverna
      device_class: window
      disabled_by_default: False
    - id: zone_3
      name: Sensore DT Garage
      device_class: motion
      disabled_by_default: False
    - id: zone_4
      name: Scuretto Finestra Lavanderia
      device_class: window
      disabled_by_default: False
    - id: zona_1_sabotaggio
      name: Sabot. Sensore DT Taverna
      disabled_by_default: False
    - id: zona_2_sabotaggio
      name: Sabot. Finestra Taverna
      disabled_by_default: False
    - id: zona_3_sabotaggio
      name: Sabot. Sensore DT Garage
      disabled_by_default: False
    - id: zona_4_sabotaggio
      name: Sabot. Scuretto Finestra Lavanderia
      disabled_by_default: False
    - id: excluded_zone_1
      name: Escl. Sensore DT Taverna
      disabled_by_default: False
    - id: excluded_zone_2
      name: Escl. Finestra Taverna
      disabled_by_default: False
    - id: excluded_zone_3
      name: Escl. Sensore DT Garage
      disabled_by_default: False
    - id: excluded_zone_4
      name: Escl. Scuretto Finestra Lavanderia
      disabled_by_default: False
    - id: alarm_memory_zone_1
      name: Mem. Allarme Sensore DT Taverna
      disabled_by_default: False
    - id: alarm_memory_zone_2
      name: Mem. Allarme Finestra Taverna
      disabled_by_default: False
    - id: alarm_memory_zone_3
      name: Mem. Allarme Sensore DT Garage
      disabled_by_default: False
    - id: alarm_memory_zone_4
      name: Mem. Allarme Scuretto Finestra Lavanderia
      disabled_by_default: False
    - id: sabotage_memory_zone_1
      name: Mem. Sabotaggio Sensore DT Taverna
      disabled_by_default: False
    - id: sabotage_memory_zone_2
      name: Mem. Sabotaggio Finestra Taverna
      disabled_by_default: False
    - id: sabotage_memory_zone_3
      name: Mem. Sabotaggio Sensore DT Garage
      disabled_by_default: False
    - id: sabotage_memory_zone_4
      name: Mem. Sabotaggio Scuretto Finestra Lavanderia
      disabled_by_default: False
    - id: alarm_area_1
      name: Allarme Finestre
      disabled_by_default: False
    - id: alarm_area_2
      name: Allarme Area 2
      disabled_by_default: False
    - id: alarm_area_3
      name: Allarme Area 3
      disabled_by_default: False
    - id: alarm_area_4
      name: Allarme Area Sensori Interni
      disabled_by_default: False
    - id: enabled_total_area_1
      name: Ins. Totale Finestre
      disabled_by_default: False
    - id: enabled_total_area_2
      name: Ins. Totale Porte
      disabled_by_default: False
    - id: enabled_total_area_3
      name: Ins. Totale Sensori Giardino
      disabled_by_default: False
    - id: enabled_total_area_4
      name: Ins. Totale Sensori Interni
      disabled_by_default: False
    - id: enabled_partial_area_1
      name: Ins. Parziale Finestre
      disabled_by_default: False
    - id: enabled_partial_area_2
      name: Ins. Parziale Porte
      disabled_by_default: False
    - id: enabled_partial_area_3
      name: Ins. Parziale Sensori Giardino
      disabled_by_default: False
    - id: enabled_partial_area_4
      name: Ins. Parziale Sensori Interni
      disabled_by_default: False
    - id: enabled_partial_delay_0_area_1
      name: Ins. Parziale Ritardo 0 Finestre
      disabled_by_default: False
    - id: enabled_partial_delay_0_area_2
      name: Ins. Parziale Ritardo 0 Porte
      disabled_by_default: False
    - id: enabled_partial_delay_0_area_3
      name: Ins. Parziale Ritardo 0 Sensori Giardino
      disabled_by_default: False
    - id: enabled_partial_delay_0_area_4
      name: Ins. Parziale Ritardo 0 Sensori Interni
      disabled_by_default: False
    - id: disabled_area_1
      name: Disins. Finestre
      disabled_by_default: False
    - id: disabled_area_2
      name: Disins. Porte
      disabled_by_default: False
    - id: disabled_area_3
      name: Disins. Sensori Giardino
      disabled_by_default: False
    - id: disabled_area_4
      name: Disins. Sensori Interni
      disabled_by_default: False
    - id: exit_status_1
      name: Stato Uscita 1
      disabled_by_default: False
    - 
      ID exit_status_1 redefined! Check binary_sensor->0->binary_sensors->55->id.
      id: exit_status_1
      name: Stato Uscita 1
      disabled_by_default: False
    - 
      ID exit_status_1 redefined! Check binary_sensor->0->binary_sensors->55->id.
      id: exit_status_1
      name: Stato Uscita 1
      disabled_by_default: False
    - 
      ID exit_status_1 redefined! Check binary_sensor->0->binary_sensors->55->id.
      id: exit_status_1
      name: Stato Uscita 1
      disabled_by_default: False`

lorenzo-deluca, 2023-10-29:
Ciao @fabix683,
a prima vista sembra che hai duplicato "exit_status_1"
Fai un check, eventualmente prova a postare l'intero yaml (senza password, ovviamente).

Fammi sapere come va ;)

fabix683, 2023-10-29:
Grazie per la risposta. 
Ho usato come file di configurazione quello allegato al progetto ed in effetti ho visto che alla fine "exit_status_1" è duplicato 4 volte
Ho corretto questo errore ma ancora qualcosa non va, ora ricevo 

Compiling .pioenvs/espkyogatebox/libfe8/ESPAsyncTCP-esphome/AsyncPrinter.cpp.o
/config/esphome/esphome-web-53c388.yaml: In lambda function:
/config/esphome/esphome-web-53c388.yaml:115:56: error: expected primary-expression before '.' token
  115 |         &kyo32->inserimento_parziale_ritardo_0_area[3],.
      |                                                        ^
/config/esphome/esphome-web-53c388.yaml:117:9: error: expected unqualified-id before '&' token
  117 |         &kyo32->disinserita_area[0],

Allego file completo del log e file Yaml usato 
[logs_esphome-web-53c388_compile (1).txt](https://github.com/lorenzo-deluca/espkyogate/files/13197690/logs_esphome-web-53c388_compile.1.txt)

lorenzo-deluca, 2023-10-29:
Linea 115 del file yaml, c'è un "." di troppo ;)

closed 2023-10-29 by lorenzo-deluca