lorenzo-deluca/espkyogate #46: Connessione API

serucci, 2024-01-05:
Buongiorno ,al posto della semplice pswd per accedere al modulo esp tramite api, vorrei implementare una encryption key . Il problema è che una volta compilato il firmware e flashato sul modulo esp , se da home assistant con l'add-on esphome cerco di visualizzare il log del modulo esp ,ricevo questo errore :
```
INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/espkyogate_configuration.yaml...
INFO Starting log output from 192.168.1.26 using esphome API
WARNING Can't connect to ESPHome API for espkyogatebox @ 192.168.1.26: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.26', port=6053))]: [Errno 111] Connect call failed ('192.168.1.26', 6053) (SocketAPIError)
INFO Trying to connect to espkyogatebox @ 192.168.1.26 in the background
```
non riesco a capire quale sia il problema per cui pare non esponga le api, puoi aiutarmi ?

lorenzo-deluca, 2024-01-05:
Ciao @serucci , mmmh, onestamente non ho mai usato la encryption key ma è a livello di ESPHome quindi non penso sia un problema della mia integrazione giusto?
Con la semplice password funziona no?
Eventualmente dai un occhio alla documentazione di ESPHome e fammi sapere come configurare con encryption key oltre alla password ;)

closed 2024-01-05 by lorenzo-deluca

serucci, 2024-01-05:
Non so se è un problema della tua integrazione , tu nel file .yaml la richiedi ed è uno dei 2 modi di impostarla, l'altro e mettere direttamente la encryption e la pswd senza passare dal secret file

```
# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password
```

con la semplice pswd da lo stesso errore

```
# Enable Home Assistant API
api:
  password: pluto

ota:
  password: pippo
```

oltretutto home assistant  mi richiede una riparazione sul modulo espkyogate

```
La password API per ESPHome è obsoleta e si consiglia invece l'uso di una chiave di crittografia API.

Rimuovi la password dell'API e aggiungi una chiave di crittografia al tuo dispositivo ESPHome per risolvere questo problema  
```

lorenzo-deluca, 2024-01-05:
La riparazione sarà un warning perché consigliano l'encryption key , ma non è mandatorio.

In ogni caso con questa configurazione funziona a tutti, l'ho appena ricompilata con ESPHome 2023.12.5 e funziona perfettamente.

```yaml
# Enable Home Assistant API
api:
  password: !secret ha_password

ota:
  password: !secret ota_password
```

Il tuo errore è dovuto al fatto che non riesce nemmeno a collegarsi, non credo proprio dipenda dalla mia integrazione questo.
Hai provato a riflashare collegando la scheda via USB?

In ogni caso la password è meglio che sia definita anche su ESPHome così è già tutto comply.

serucci, 2024-01-05:
compilare compila perfettamente e il flash lo faccio collegando il modulo via usb ,è quando cerco di vedere il log che da quell'errore

lorenzo-deluca, 2024-01-05:
Si ma da quell'errore sembra che non riesca neppure a collegarsi al wifi..
al ping quell'ip risponde?

-- 

Regards,
*Lorenzo De Luca*

***@***.***
***@***.***>
*DevOps & **CTO*

dddprint.io

*PRIVACY INFORMATION*
The information contained in this e-mail message and/or in the enclosed
file/s is/are strictly
reserved. Their use is only allowed the addressee of the message, for the
purposes showed in the message itself. If you receive this message without
being its addressee, we kindly ask You to inform us about it by e-mail and
to proceed to the destruction of the same message cancelling it from Your
system.
It constitutes opposite behaviour to the principles prescribed by the
Legislative Decree 196/2003 the conservation of the message, the
divulgation and distribution of that message to other subjects, the copy of
it, or the use of it for different purposes.

P  Respects nature.
Do not print this email unless absolutely necessary.

serucci, 2024-01-05:
si risponde

lcavalli, 2024-01-05:
La compilazione ha dato worning o errori? Hai provato un clean prima di ricompilare? L'errore potrebbe essere dovuto al fatto che l'esp continui a crashare e resettarsi. Se hai modo col cavo USB dovresti riuscire a raccogliere più informazioni.

serucci, 2024-01-05:
Nessun errore a fine compilazione e quando faccio flash sono collegato con usb....com si fa il clean prima di compilare? io vedo che il sw flashesp8266 prima di scrivere fa un erasing flash

lorenzo-deluca, 2024-01-05:
Grazie Luca :) 
Il clean penso si riferisca alla build di compilazione non tanto all'erase della memoria dell'ESP.
Comunque a questo punto proverei a cambiare scheda ESP, magari passato a ESP32 che è più performante anche nella gestione della comunicazione seriale, etc..

serucci, 2024-01-05:
penso che farò cosi...grazie per la disponibilità , ciao buona serata