lorenzo-deluca/espkyogate #49: KYO8 ed attivazione allarme

pow3r77, 2024-02-11:
Ciao Lorenzo,
complimenti per il lavoro, lo sto testando su una centrale KYO8
 Quando lancio il servizio per armare la zona 1 (unica zone nella mia centrale configurata) la zona si attiva correttamente ma partono anche le chiamate telefoniche di attivazione allarme. Stessa cosa quando disarmo con il service "disarm"

tutte le volte sento scattare un "doppio" relè e si attiva il compositore telefonico.

Hai qualche idea?

Grazie mille

lorenzo-deluca, 2024-02-11:
Ciao, non ho mai avuto il compositore telefonico.. però è strano, nessuno mi ha mai segnalato questo problema.
Hai provato anche con l'abilitazione parziale?

pow3r77, 2024-02-11:
In pratica attivando con la modalità 1 mi fa un "attivazione speciale" che dovrebbe corrispondere all'attivazione "antipanico"in affetto, motivo per il quale fa partire il combinatore

[image]

Adesso provo altri tipi di attivazione

pow3r77, 2024-02-11:
Allora, ho inviato ex-novo la configurazione e adesso funziona tutto. Ovviamente non trovo un nesso logico, ma non mi attiva in automatico le telefonate.
Ritieni ci sia un modo per chiamare anche il servizio di "cancella telefonate" ? Essendo una delle opzioni del menu quando si attiva il combinatore. Sarebbe molto utile per interrompere il "loop" delle chiamate

pow3r77, 2024-02-12:
Giusto per darti un feedback generale, su KYO8 probabilmente alcuni voci non sono allineate, quando attivo allarme totale area 1, mi si attiva il sensore " - id: disabled_area_1"
Se faccio un attivazione della medesima area parziale non si attiva invece nessuno dei sensori creati in HA.
Ritieni ci sia possibilità di allinearli?
Grazie infinite della disponibilità, qualora non avessi tempo e voglia di seguire le mie indicazioni comprendo ;)
Grazie e buona serata

lorenzo-deluca, 2024-02-12:
Ciao, grazie del feedback.
Penso possa essere un problema di configurazione yaml perché mi hanno scritto diversi utenti con Kyo 8G, 8M e 8W che non hanno riscontrato problemi.
Puoi postare qui la tua configurazione yaml (senza password, ovviamente)?

pow3r77, 2024-02-12:
esphome:
  name: espkyogatebox
  friendly_name: Allarme Casa
  comment: Alarm System Serial to HA controller
  includes:
    - bentel-kyo/bentel_kyo32.h

esp8266:
  board: esp01_1m

# Enable Home Assistant API
api:
  encryption:
    key: "uJxOXuy3xy1LmWQvHPtoUAQzOfuhIIP/IeuH761viqU="

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-2C8709"
    password: "4IYsoV6qPx0E"

captive_portal:

# Enable logging
logger:
  level: INFO
  baud_rate: 0
  logs:
    component: ERROR

  
uart:
  id: uart_bus
  tx_pin: GPIO5
  rx_pin: GPIO4
  baud_rate: 9600
  data_bits: 8
  parity: EVEN
  
switch:
  - platform: safe_mode
    name: "ESPKyoGate (Safe Mode)"

binary_sensor:
  - platform: custom
    lambda: |-
      auto kyo32 = new Bentel_Kyo32(id(uart_bus));
      App.register_component(kyo32);
      return {
        kyo32->kyo_comunication,
        kyo32->stato_sirena,
        kyo32->warn_mancanza_rete,
        kyo32->sabotaggio_zona, 
        kyo32->sabotaggio_sistema,
        kyo32->warn_scomparsa_bpi,
        kyo32->warn_fusibile,
        kyo32->warn_batteria_bassa,
        kyo32->warn_guasto_linea_telefonica,
        kyo32->warn_codici_default,
        kyo32->warn_wireless,
        kyo32->sabotaggio_bpi,
        kyo32->sabotaggio_jam,
        kyo32->sabotaggio_wireless,
        kyo32->sabotaggio_chiave_falsa,

[... 167 lines cut ...]
      #- id: enabled_partial_area_3
      #  name: "Ins. Parziale Sensori Giardino"
      #- id: enabled_partial_area_4
      #  name: "Ins. Parziale Sensori Interni"        

      - id: enabled_partial_delay_0_area_1
        name: "Ins. Parziale Ritardo 0 Casa"
      #- id: enabled_partial_delay_0_area_2
      #  name: "Ins. Parziale Ritardo 0 Porte"
      #- id: enabled_partial_delay_0_area_3
      #  name: "Ins. Parziale Ritardo 0 Sensori Giardino"
      #- id: enabled_partial_delay_0_area_4
      #  name: "Ins. Parziale Ritardo 0 Sensori Interni"

      - id: disabled_area_1
        name: "Allarme Totale"
      - id: disabled_area_2
        name: "Allarme Parziale"
      #- id: disabled_area_3
      #  name: "Disins. Sensori Giardino"
      #- id: disabled_area_4
      #  name: "Disins. Sensori Interni"
      
      - id: exit_status_1
        name: "Stato Uscita 1"

pow3r77, 2024-02-12:
`esphome:
  name: espkyogatebox
  friendly_name: Allarme Casa
  comment: Alarm System Serial to HA controller
  includes:
    - bentel-kyo/bentel_kyo32.h

esp8266:
  board: esp01_1m

# Enable Home Assistant API
api:
  encryption:
    key: "uJxOXuy3xy1LmWQvHPtoUAQzOfuhIIP/IeuH761viqU="

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-2C8709"
    password: "4IYsoV6qPx0E"

captive_portal:

# Enable logging
logger:
  level: INFO
  baud_rate: 0
  logs:
    component: ERROR

  
uart:
  id: uart_bus
  tx_pin: GPIO5
  rx_pin: GPIO4
  baud_rate: 9600
  data_bits: 8
  parity: EVEN
  
switch:
  - platform: safe_mode
    name: "ESPKyoGate (Safe Mode)"

binary_sensor:
  - platform: custom
    lambda: |-
      auto kyo32 = new Bentel_Kyo32(id(uart_bus));
      App.register_component(kyo32);
      return {
        kyo32->kyo_comunication,
        kyo32->stato_sirena,
        kyo32->warn_mancanza_rete,
        kyo32->sabotaggio_zona, 
        kyo32->sabotaggio_sistema,
        kyo32->warn_scomparsa_bpi,
        kyo32->warn_fusibile,
        kyo32->warn_batteria_bassa,
        kyo32->warn_guasto_linea_telefonica,
        kyo32->warn_codici_default,
        kyo32->warn_wireless,
        kyo32->sabotaggio_bpi,
        kyo32->sabotaggio_jam,
        kyo32->sabotaggio_wireless,
        kyo32->sabotaggio_chiave_falsa,

[... 167 lines cut ...]
      #- id: enabled_partial_area_3
      #  name: "Ins. Parziale Sensori Giardino"
      #- id: enabled_partial_area_4
      #  name: "Ins. Parziale Sensori Interni"        

      - id: enabled_partial_delay_0_area_1
        name: "Ins. Parziale Ritardo 0 Casa"
      #- id: enabled_partial_delay_0_area_2
      #  name: "Ins. Parziale Ritardo 0 Porte"
      #- id: enabled_partial_delay_0_area_3
      #  name: "Ins. Parziale Ritardo 0 Sensori Giardino"
      #- id: enabled_partial_delay_0_area_4
      #  name: "Ins. Parziale Ritardo 0 Sensori Interni"

      - id: disabled_area_1
        name: "Allarme Totale"
      - id: disabled_area_2
        name: "Allarme Parziale"
      #- id: disabled_area_3
      #  name: "Disins. Sensori Giardino"
      #- id: disabled_area_4
      #  name: "Disins. Sensori Interni"
      
      - id: exit_status_1
        name: "Stato Uscita 1"    `

pow3r77, 2024-02-12:
```
esphome:
  name: espkyogatebox
  friendly_name: Allarme Casa
  comment: Alarm System Serial to HA controller
  includes:
    - bentel-kyo/bentel_kyo32.h

esp8266:
  board: esp01_1m

# Enable Home Assistant API
api:
  encryption:
    key: "uJxOXuy3xy1LmWQvHPtoUAQzOfuhIIP/IeuH761viqU="

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-2C8709"
    password: "4IYsoV6qPx0E"

captive_portal:

# Enable logging
logger:
  level: INFO
  baud_rate: 0
  logs:
    component: ERROR

  
uart:
  id: uart_bus
  tx_pin: GPIO5
  rx_pin: GPIO4
  baud_rate: 9600
  data_bits: 8
  parity: EVEN
  
switch:
  - platform: safe_mode
    name: "ESPKyoGate (Safe Mode)"

binary_sensor:
  - platform: custom
    lambda: |-
      auto kyo32 = new Bentel_Kyo32(id(uart_bus));
      App.register_component(kyo32);
      return {
        kyo32->kyo_comunication,
        kyo32->stato_sirena,
        kyo32->warn_mancanza_rete,
        kyo32->sabotaggio_zona, 
        kyo32->sabotaggio_sistema,
        kyo32->warn_scomparsa_bpi,
        kyo32->warn_fusibile,
        kyo32->warn_batteria_bassa,
        kyo32->warn_guasto_linea_telefonica,
        kyo32->warn_codici_default,
        kyo32->warn_wireless,
        kyo32->sabotaggio_bpi,
        kyo32->sabotaggio_jam,
        kyo32->sabotaggio_wireless,
        kyo32->sabotaggio_chiave_falsa,
[... 169 lines cut ...]
      #  name: "Ins. Parziale Sensori Giardino"
      #- id: enabled_partial_area_4
      #  name: "Ins. Parziale Sensori Interni"        

      - id: enabled_partial_delay_0_area_1
        name: "Ins. Parziale Ritardo 0 Casa"
      #- id: enabled_partial_delay_0_area_2
      #  name: "Ins. Parziale Ritardo 0 Porte"
      #- id: enabled_partial_delay_0_area_3
      #  name: "Ins. Parziale Ritardo 0 Sensori Giardino"
      #- id: enabled_partial_delay_0_area_4
      #  name: "Ins. Parziale Ritardo 0 Sensori Interni"

      - id: disabled_area_1
        name: "Allarme Totale"
      - id: disabled_area_2
        name: "Allarme Parziale"
      #- id: disabled_area_3
      #  name: "Disins. Sensori Giardino"
      #- id: disabled_area_4
      #  name: "Disins. Sensori Interni"
      
      - id: exit_status_1
        name: "Stato Uscita 1"    
```

lorenzo-deluca, 2024-02-12:
Ciao, allora, confermo che c'è sicuramente un problema di configurazione.
Fai attenzione, le variabili mappate nel return DEVONO essere mappate uno a uno (in esatto ordine) con le variabili sotto.

Nella tua configurazione già vedo che le zone ritornate sono 4 (da 0 a 3), mentre quelle mappate sono 5, idem per le zone di sabotaggio (allego screen).
Sfortunatamente, essendo tutti binary sensors, se il numero è uguale funziona lo stesso ma ovviamente ti va a mappare logicamente valori diversi rispetto a quanto ti aspetti.

[image]

In futuro vedrò se riesco a trovare un mapping più semplice perché non sei il primo che cade in questo errore 👍 

Fammi sapere come va ;)

pow3r77, 2024-02-12:
Perfetto! Grazie mille per le spiegazioni! Adesso ho sistemato tutto e mi tornano tutti i comandi.
Riguardo alla possibilità di inserire il comando per interrompere le chiamate? Ritieni si possa riuscire?

lorenzo-deluca, 2024-02-13:
Ottimo, mi sembrava strano che solo tu avessi problemi.
Per il comando di reset chiamate il reset allarme non basta vero?
Il fatto è che non avendo il compositore mi dovete dire voi come funziona 😅 fondamentalmente, dal software Bentel dell'installatore si può fare quel comando? Se si come?

closed 2024-02-13 by lorenzo-deluca

pow3r77, 2024-02-13:
Dal software bentel non mi pare di aver, lo puoi fare come comando dalla centrale.

https://www.youtube.com/watch?v=proCnc0EoHE