lorenzo-deluca/espkyogate #50: Allarme aree e uscite non si attivano

serucci, 2024-02-14:
Ciao Lorenzo ,dopo avere caricato nel modulo ESP il nuovo firmware ci sono 2 cose che non mi funzionano. Presumo che sia un problema di configurazione ma non riesco a trovare l'inghippo. 
I binary sensor dell'allarme aree non si attivano mai , nessuno delle 8 aree. lo stesso dicasi di quelli dello stato delle uscite
Questo è il file .yaml della configurazione
Qualche idea? grazie
```
esphome:
  name: espkyogatebox
  comment: Alarm System Serial to HA controller
  platform: ESP8266
  board: d1_mini
  includes:
    - EspkyogateBoxOk/bentel_kyo32.h

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: !secret hotspot_wifi_ssid
    password: !secret hotspot_wifi_password

captive_portal:

# Enable logging
logger:
  level: VERBOSE
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password
  
uart:
  id: uart_bus
  tx_pin: GPIO5
  rx_pin: GPIO4
  baud_rate: 9600
  data_bits: 8
  parity: EVEN
  
switch:
  - platform: safe_mode
    name: "ESPKyoGate (Safe Mode)"

binary_sensor:
  - platform: custom
    lambda: |-
      auto kyo32 = new Bentel_Kyo32(id(uart_bus));
      App.register_component(kyo32);
      return {
        kyo32->kyo_comunication,
        kyo32->stato_sirena,
        kyo32->warn_mancanza_rete,
        kyo32->sabotaggio_zona, 
        kyo32->sabotaggio_sistema,
        kyo32->warn_scomparsa_bpi,
        kyo32->warn_fusibile,
        kyo32->warn_batteria_bassa,
        kyo32->warn_guasto_linea_telefonica,
        kyo32->warn_codici_default,
        kyo32->warn_wireless,
        kyo32->sabotaggio_bpi,
        kyo32->sabotaggio_jam,
        kyo32->sabotaggio_wireless,
        kyo32->sabotaggio_chiave_falsa,
[... 421 lines cut ...]
        name: "Disins. FinestreCortile"
      - id: disinserita_area_3
        name: "Disins. VolumetricoTinello"
      - id: disinserita_area_4
        name: "Disins. TapparelleStrada"
      - id: disinserita_area_5
        name: "Disins. FinestreStrada"
      - id: disinserita_area_6
        name: "Disins. VolumetricoFrancy"
      - id: disinserita_area_7
        name: "Disins. VolumetricoIngresso"
      - id: disinserita_area_8
        name: "Disins. PortaBlindata"    

      - id: stato_uscita_1
        name: "Stato Uscita 1"
      - id: stato_uscita_2
        name: "Stato Uscita 2"
      - id: stato_uscita_3
        name: "Stato Uscita 3"
      - id: stato_uscita_4
        name: "Stato Uscita 4"
      - id: stato_uscita_5
        name: "Stato Uscita 5"
```

lorenzo-deluca, 2024-02-17:
Ciao, mi sembra tutto ok.. che centrale hai, Kyo 32?
Prima dell'ultimo aggiornamento dell'ESP i sensori si attivavano tutti correttamente?

Io proverei a riflashare il firmware...

serucci, 2024-02-17:
Ciao ,la centrale è una Kyo32g fino a qualche mese fa ho usato la prima versione del tuo firmware, già allora eravamo incappati nel problema del reset allarmi che non funzionava su tutte le zone e sui binary delle aree che non cambiavano di stato. Poi hai sistemato il tutto e da li è tutto funzionato a meraviglia . Ora ho aggiornato il tutto per utilizzare la encryption key per l'accesso alle api. Il firmware l'ho riflashato più volte ,anche provando ad utilizzare l'originale senza aggiungere nulla, ma nulla cambia. 
A te funziona tutto ?

serucci, 2024-02-21:
Aggiungo che, anche provando ad utilizzare l'ultimo firmware senza aggiungere o togliere nessun binary sensor ,per adattarlo alle mie esigenze, i binary allarme_area[x] non cambiano mai di stato , idem per i binary stato_uscita[x]
Mi piacerebbe capire se altri hanno testato tutti i singoli binary_sensor e se gli funzionano o sbaglio qualcosa io
Nel post precedente ti ho erroneamente scritto che nella prima versione del fw gli allarme area funzionavano, non è vero....

lorenzo-deluca, 2024-04-17:
Ciao a tutti, effettivamente noto anche io che lo stato delle uscite viene visto come sconosciuto.
Dovrò fare un'analisi con la centrale di test che avevo in giro, sperando funzioni ancora, però sono abbastanza sicuro che una volta funzionavano .... ovviamente non ricordo quando, per caso qualche altro utente ricorda se questa segnalazione di stato funzionasse?

bassguitarist, 2024-04-17:
se posso aiutare.... la mia centrale non è ancora operativa, ci sto mettendo un po' più del previsto a tirar cavi e capire come programmarla. Quindi se vuoi farmi fare qualche test sono a disposizione

serucci, 2024-04-17:
> Ciao a tutti, effettivamente noto anche io che lo stato delle uscite viene visto come sconosciuto. Dovrò fare un'analisi con la centrale di test che avevo in giro, sperando funzioni ancora, però [...]

Ciao Lorenzo, lo stato delle uscite non è mai funzionato, quello che andava con la prima versione del firmware erano i binary degli allarmi_area

lorenzo-deluca, 2024-04-17:
Ciao ragazzi, dovrei aver risolto! 👍 
c'era un piccolo problema, con la Kyo32 dovrebbe andare senza problemi, andrebbe verificato sulla Kyo8.

Fatemi sapere se vi torna tutto... grazie della segnalazione

closed 2024-04-17 by lorenzo-deluca

bassguitarist, 2024-04-17:
Lorenzo... grazie mille!
Confermo che il codice aggiornato funziona!

Ecco ad esempio la mia "uscita 1" attivata in concomitanza alla sirena
[image]

Buona serata

serucci, 2024-04-19:
Confermo il corretto funzionamento dello stato delle uscite , mentre ancora i binary sensor allarmi area rimangono sempre spenti
Comunque grazie Lorenzo

lorenzo-deluca, 2024-04-19:
> Confermo il corretto funzionamento dello stato delle uscite , mentre ancora i binary sensor allarmi area rimangono sempre spenti Comunque grazie Lorenzo

Prego, un po alla volta vedo di sistemare tutti i problemini...
Se hai voglia apri pure un'altra Issue per lo stato binary sensor allarmi, ho visto solo ora quella dell'antisabotaggio (che io ho sempre bypassato).
Quando metto in piedi la centrale di test volevo anche aggiungere il comando sirena che potrebbe servire anche a me, vedo di sistemare anche questo.

Ciao :)

serucci, 2024-04-21:
> Prego, un po alla volta vedo di sistemare tutti i problemini... Se hai voglia apri pure un'altra Issue per lo stato binary sensor allarmi, ho visto solo ora quella dell'antisabotaggio (che io ho [...]

Te la butto li, visto che metterai in piedi la centrale di test ,c'era un altro utente che come me a suo tempo, ti chiedeva la possibilità di poter bloccare le telefonate in arrivo dalla centrale. Anche se non hai il combinatore telefonico ,penso che tu possa leggere il comando inviato dalla tastiera alla centrale , menu utente, funzione "Canc. Telefonate" ...al tuo buon cuore :)) ciao

lorenzo-deluca, 2024-04-21:
Si, ricordo che era già stato chiesto.
Il fatto è che io vedo solo la comunicazione con il software dell'installatore, se si può fare anche da li posso vedere il comando altrimenti da tastiera non vedo nulla. 
Tu hai il software installatore hai provato a fare il cancella chiamate?

Comunque quando metto in piedi la centrale di test faccio qualche prova.

serucci, 2024-04-21:
> Si, ricordo che era già stato chiesto. Il fatto è che io vedo solo la comunicazione con il software dell'installatore, se si può fare anche da li posso vedere il comando altrimenti da tastiera non [...]

Se per software installatore intendi bentel security suite , si ce l'ho ,ci ho appena guardato ma non mi pare che si possa fare la cancellazione delle chiamate ,almeno, io non vedo nulla