lorenzo-deluca/espkyogate #57: stato binary sensor allarmi

serucci, 2024-04-20:
Come già scritto nella issue chiusa ,non cambiano di stato...ciao

serucci, 2024-04-20:
con molto piacere caffe...offerti

newtoni85, 2024-04-20:
Buongiorno, non sono esperto di esphome.... nella mia configurazione i sensori allarme area funzionano.
[image]

lorenzo-deluca, 2024-04-20:
> Buongiorno, non sono esperto di esphome.... nella mia configurazione i sensori allarme area funzionano. <img alt="Schermata 2024-04-20 alle 11 25 57" width="695" src="https://private-user-images.githubusercontent.com/123250397/324168356-94b6266e-5cfa-4e43-b834-194fb316babe.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM2MDU5OTksIm5iZiI6MTcxMzYwNTY5OSwicGF0aCI6Ii8xMjMyNTAzOTcvMzI0MTY4MzU2LTk0YjYyNjZlLTVjZmEtNGU0My1iODM0LTE5NGZiMzE2YmFiZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyMFQwOTM0NTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04Y2I3YTU0M2FjODM1OTZlODVkMTZiNzk1ZTk0NWU0YzU5ZTQ5YTZhY2I1OTRhNTMzNjI2NjJkNTQ0YjM5ZjExJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.MbVkf7U3F33RI5Y9g-cyhaS9y-4hhsRVyZY0MdvnM3M">

Infatti anche io ricordavo funzionassero.. stai usando l'ultima versione? la centrale è una Kyo8 o Kyo32? Puoi postare la tua configurazione yaml (senza password, ovviamente)..

Grazie!

newtoni85, 2024-04-20:
> Infatti anche io ricordavo funzionassero.. stai usando l'ultima versione? la centrale è una Kyo8 o Kyo32? Puoi postare la tua configurazione yaml (senza password, ovviamente)..
>
> Grazie!

la centrale è una Kyo32G e attulamente il file bentel_kyo32.h utilizzato è la precedente versione a quella che hai pubblicato tre giorni fa. per la configurazione nessun problema come funziona posso inserita nei commenti?

newtoni85, 2024-04-20:
esphome:
  name: bentel
  friendly_name: bentel
  includes:
    - build/bentel/bentel_kyo32.h

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Enable logging
logger:
  level: info
  baud_rate: 0

uart:
  id: uart_bus
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 9600
  data_bits: 8
  parity: EVEN

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  password: 

wifi:
  ssid: 
  password:

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: 
    password: 

captive_portal:
  
switch:

  - platform: safe_mode
    name: "ESPKyoGate (Safe Mode)"

binary_sensor:
- platform: custom
  lambda: |-
      auto kyo32 = new Bentel_Kyo32(id(uart_bus));
      App.register_component(kyo32);
      return {
        kyo32->kyo_comunication,
        kyo32->stato_sirena,
        kyo32->warn_batteria_bassa,
        &kyo32->zona[0],
        &kyo32->zona[1],
        &kyo32->zona[2],
        &kyo32->zona[3],
        &kyo32->zona[4],
        &kyo32->zona[5],
        &kyo32->zona[6],
        &kyo32->zona[7],
        &kyo32->zona[8],
        &kyo32->zona[9],
        &kyo32->zona[10],
        &kyo32->zona[11],
        &kyo32->zona[12],
[... 167 lines cut ...]
      - id: inserimento_totale_area_2
        name: "Inserimento Totale Area 2"
      - id: inserimento_totale_area_3
        name: "Inserimento Totale Area 3"
      - id: inserimento_totale_area_4
        name: "Inserimento Totale Area 4"
      - id: inserimento_totale_area_5
        name: "Inserimento Totale Area 5"
      - id: inserimento_totale_area_6
        name: "Inserimento Totale Area 6"

      - id: disinserita_area_0
        name: "Disinserita Area 0"
      - id: disinserita_area_1
        name: "Disinserita Area 1"
      - id: disinserita_area_2
        name: "Disinserita Area 2"
      - id: disinserita_area_3
        name: "Disinserita Area 3"
      - id: disinserita_area_4
        name: "Disinserita Area 4"
      - id: disinserita_area_5
        name: "Disinserita Area 5"
      - id: disinserita_area_6
        name: "Disinserita Area 6"

lorenzo-deluca, 2024-04-20:
Grazie @newtoni85 , (ho editato il commento per la formattazione yaml)
Effettivamente anche io ricordo che funzionano.

@serucci puoi verificare la tua configurazione con quella di @newtoni85 ?

Grazie a tutti 👍

serucci, 2024-04-20:
questo è il mio yaml ,al volo vedo solo la nomenclatura nei register_component e nei binary delle aree ,sono numerate da 0 a 7. Nel mio sono 0-7 nei register_component e nei binary 1-8 , ma anche mettendoli uguali non cambia il risultato...altro non vedo

```
esphome:
  name: espkyogatebox
  comment: Alarm System Serial to HA controller
  platform: ESP8266
  board: d1_mini
  includes:
    - EspkyogateBoxOk/bentel_kyo32.h

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: !secret hotspot_wifi_ssid
    password: !secret hotspot_wifi_password

captive_portal:

# Enable logging
logger:
  level: VERBOSE
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password
  
uart:
  id: uart_bus
  tx_pin: GPIO5
  rx_pin: GPIO4
  baud_rate: 9600
  data_bits: 8
  parity: EVEN
  
switch:
  - platform: safe_mode
    name: "ESPKyoGate (Safe Mode)"

binary_sensor:
  - platform: custom
    lambda: |-
      auto kyo32 = new Bentel_Kyo32(id(uart_bus));
      App.register_component(kyo32);
      return {
        kyo32->kyo_comunication,
        kyo32->stato_sirena,
        kyo32->warn_mancanza_rete,
        kyo32->sabotaggio_zona, 
        kyo32->sabotaggio_sistema,
        kyo32->warn_scomparsa_bpi,
        kyo32->warn_fusibile,
        kyo32->warn_batteria_bassa,
        kyo32->warn_guasto_linea_telefonica,
        kyo32->warn_codici_default,
        kyo32->warn_wireless,
        kyo32->sabotaggio_bpi,
        kyo32->sabotaggio_jam,
        kyo32->sabotaggio_wireless,
        kyo32->sabotaggio_chiave_falsa,
        
        &kyo32->zona[0],
[... 420 lines cut ...]
      - id: disinserita_area_3
        name: "Disins. VolumetricoTinello"
      - id: disinserita_area_4
        name: "Disins. TapparelleStrada"
      - id: disinserita_area_5
        name: "Disins. FinestreStrada"
      - id: disinserita_area_6
        name: "Disins. VolumetricoFrancy"
      - id: disinserita_area_7
        name: "Disins. VolumetricoIngresso"
      - id: disinserita_area_8
        name: "Disins. PortaBlindata"    

      - id: stato_uscita_1
        name: "Stato Uscita 1"
      - id: stato_uscita_2
        name: "Stato Uscita 2"
      - id: stato_uscita_3
        name: "Stato Uscita 3"
      - id: stato_uscita_4
        name: "Stato Uscita 4"
      - id: stato_uscita_5
        name: "Stato Uscita 5"

```

lorenzo-deluca, 2024-04-21:
Scusa, che modulo usi tu? Dalla configurazione mi sembra di capire un esp8266.
Perché vedo che hai configurato molti sensori e più utenti mi hanno segnalato che il "povero" esp8266 fa fatica a gestire molti sensori.

Eventualmente se non hai un ESP32 o superiore puoi provare a fare una configurazione in cui mappi solo i sensori dell'allarme area?

Grazie

newtoni85, 2024-04-21:
> Scusa, che modulo usi tu? Dalla configurazione mi sembra di capire un esp8266.
> Perché vedo che hai configurato molti sensori e più utenti mi hanno segnalato che il "povero" esp8266 fa fatica a [...]

Anche io se posso essere utile ho dovuto fare con ESP32 perché con 8266 mi dava problemi di collegamento.

serucci, 2024-04-21:
> Scusa, che modulo usi tu? Dalla configurazione mi sembra di capire un esp8266. Perché vedo che hai configurato molti sensori e più utenti mi hanno segnalato che il "povero" esp8266 fa fatica a [...]
Uso un ESP8266 e i vari sensori , 15 zone, sono raggruppati/divisi per fare intervenire le varie aree ,8 .
Comunque si ,posso provare a ridurre il numero  di sensori e posso anche provare con un  ESP32

lorenzo-deluca, 2024-06-10:
E' stato risolto il problema con l'ultima versione di firmware e aggiornamento a ESP32?

closed 2024-06-10 by lorenzo-deluca

serucci, 2024-06-10:
No ,non ho risolto ne con l'ultima versione del fw ne riducendo il numero
dei sensori...mi rimane solo l'uso del ESP32