lorenzo-deluca/espkyogate #58: Impossibilità ad effettuare l'OTA

amastrogiacomo1968, 2024-06-20:
Ho appena aggiornarnato ESPHome all'ultima versione 2024.6.1, quando vorrei aggiornare il mio esp ho questo errore

`INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/antifurto-esp32.yaml...
Failed config

At least one platform must be specified for 'ota'; add 'platform: esphome' for original OTA functionality`.

Mi sono "perso" qualcosa da fare?

Grazie in anticipo

lorenzo-deluca, 2024-06-20:
Ciao, sta scritto nelle breaking change di esphome [https://next.esphome.io/changelog/2024.6.0.html#ota-platforms](https://next.esphome.io/changelog/2024.6.0.html#ota-platforms)

Comunque aggiorno il file di configurazione di esempio.

closed 2024-06-20 by lorenzo-deluca

amastrogiacomo1968, 2024-06-20:
Ecco me lo ero perso, grazie