lorenzo-deluca/espkyogate #59: servizio Reset Alarm Memory

popeye70, 2024-06-28:
Ciao Lorenzo,
gentilmente mi confermi la sintassi da utilizzare per chiamare il servizio:
_service: esphome.espkyogate_reset_alarms_
affinchè cancelli gli errori in memoria di tutte le zone (io ne ho cinque) esistenti?

Al momento ho pedestremente copiato quanto indicato nel manuale:
_service: esphome.espkyogate_reset_alarms
data: {}_
ma sembrerebbe non funzionare...

Grazie mille
Flavio

serucci, 2024-07-13:
Ciao, confermo la sintassi corretta
```
alias: Reset Memoria Allarme
sequence:
  - delay:
      hours: 0
      minutes: 0
      seconds: 1
      milliseconds: 0
  - service: esphome.espkyogatebox_reset_alarms
    data: {}
  - delay:
      hours: 0
      minutes: 0
      seconds: 1
      milliseconds: 0
  - service: notify.xxxx
    data:
      message: "Inviato comando reset memorie allarmi "
mode: single
```

popeye70, 2024-07-18:
Ciao serucci.
cercando di salvare lo script che hai suggerito ottengo l'errore:

Message malformed: extra keys not allowed @ data['sequence'][0]['mode']

Hai altri ssuggerimenti?
Grazie

closed 2025-01-27 by lorenzo-deluca

popeye70, 2025-01-29:
Ciao, no la sintassi che mi era stata suggerita era formalmente invalida quindi ho lasciato quella che indicavi tu nelle istruzioni, anche se mi rimane il dubbio che non cancelli gli errori di tutte le aree censite ma solo della prima.Ma ci stai ancora lavorando su questo progetto? Ho visto che su Github hai fatto un pó di pulizia...

ciao, @popeye70 , hai poi risolto?

lorenzo-deluca, 2025-01-29:
Ciao  @popeye70 , si, sto lavorando con degli utenti ad una nuova versione che risolve qualche incompatibilità con le Kyo8 e Kyo4 (che non ho e mi stanno aiutando loro)
Interessante, hai verificato questa ipotesi? Se si fammi sapere, se mi mandi i log posso verificare.

Grazie, ciao!

popeye70, 2025-01-30:
Ciao Lorenzo,

ho rifatto qualche prova e ti confermo che lo script di Reset Alarm Memory funziona. Probabilmente, a suo tempo ero incappato nel medesimo problema che avevi poi risolto ad un altro utente relativo al reset allarmi delle aree dopo la terza (io ne ho cinque).

Invece, eri poi riuscito a guardare se sia possibile implementare il "batteria bassa sensore" ed eventualmente anche il "sensore scomparso"? Nel gestionale Bentel sono le info nelle due colonne a destra dell'immagine che ti allego, teoricamente dovresti quindi riuscire a prenderle così come hai fatto per gli stati di allarme e di sabotaggio.
Mi interesserebbe molto perchè sulla mia Kyo32 ho tutti sensori wireless...se ti serve un tester sono a disposizione!

Grazie e saluti.

Flavio

lorenzo-deluca, 2025-01-30:
Non vedo l'immagine allegata.. comunque io tutti gli stati che sono disponibili da seriale dovrei averli messi.

popeye70, 2025-01-30:
Provo a rimandarti la foto dal cellulare. Nello yaml io trovo gli stati di allarme, sabotaggio ed esclusione ma non quelli di batteria bassa e zona scomparsa.

popeye70, 2025-03-26:
Ciao Lorenzo scusami, che software mi consigli di usare per leggere e salvare il traffico seriale tra la kyo32 e la Bentel Suite sul PC? Grazie! Flavio