lorenzo-deluca/espkyogate #63: ESPHome 2025.2.0 e Custom Components

gianlucaf81, 2025-02-20:
So che l'aggiornamento è di ieri, ma volevo sapere se la questione verrà affrontata.
Grazie.

`INFO ESPHome 2025.2.0
INFO Reading configuration /root/config/bentel-kyo4.yaml...
Failed config

binary_sensor.custom: [source /root/config/bentel-kyo4.yaml:52]
  
  The "custom" component has been removed. Consider conversion to an external component.
  https://esphome.io/guides/contributing#a-note-about-custom-components.
  platform: custom
  lambda: |-
    auto kyo32 = new Bentel_Kyo32(id(uart_bus));
    App.register_component(kyo32);
    return {
      kyo32->kyo_comunication,
      kyo32->stato_sirena,
      kyo32->warn_mancanza_rete,
      kyo32->sabotaggio_zona, 
      kyo32->sabotaggio_sistema,
      kyo32->warn_scomparsa_bpi,`

cannos88, 2025-02-20:
io ci sto lavorando, appena ho la soluzione definitiva farò una pull request, in ogni caso si tratterebbe solo di modificare il file yaml e al massimo aggiungere un secondo file esterno insieme al .h già presente

lorenzo-deluca, 2025-02-20:
Grande @cannos88, ottimo! 👍 

Tra l'altro sarebbe anche comodo far puntare lo YAML al file .h nel direttamente dal repository GitHub, così ogni volta vi trovate la nuova versione.

adepiero, 2025-02-23:
@cannos88, oltre alla parte della platform, c'è da sistemare anche la parte iniziale dello YAML, sempre per sintassi deprecate. Sistemi tu, per favore? La nuova sintassi, almeno per le board ESP32 come la mia, è:

```
esphome:
  name: espkyogate
  friendly_name: Allarme
  comment: Alarm System Serial to HA controller
  includes:
    - components/bentel-kyo/bentel_kyo32.h

esp32:
  board: esp32doit-devkit-v1
```

lorenzo-deluca, 2025-02-24:
> [@cannos88](https://github.com/cannos88), oltre alla parte della platform, c'è da sistemare anche la parte iniziale dello YAML, sempre per sintassi deprecate. Sistemi tu, per favore? La nuova [...]

Ciao, si esatto, questo l'ho fatto anche io, tra l'altro si può far puntare più comodamente a GitHub.
Il problema mi sembra essere come rifare il mapping del binary_sensors perché è stata deprecata la "platform: custom".

@cannos88  hai fatto progressi? Se ti può essere utile possiamo guardarci anche assieme.

Grazie :)

cannos88, 2025-02-24:
ho seguito più strade ma per ora nessun progresso. Ho provato anche con il supporto di claude ai ma nada (purtroppo Gemini e chat gpt non riescono ad avere lo stesso livello di accuratezza in questo caso).

lorenzo-deluca, 2025-02-24:
> con l'occasione si può fare il merge con la versione per kyo4 oppure è troppo complesso? Grazie.
Si, una volta capito come migrare con la nuova modalità stavo pensando di fare un altro file sorgente per la Kyo4 e Kyo8 base per semplificare un po il codice.

> ho seguito più strade ma per ora nessun progresso. Ho provato anche con il supporto di claude ai ma nada (purtroppo Gemini e chat gpt non riescono ad avere lo stesso livello di accuratezza in questo caso).
Anche io ieri ho fatto qualche prova, ho trovato un modo che builda ma richiede diverse modifiche e non ho ancora provato l'effettivo funzionamento.
Se hai già un fork su cui stai lavorando fammi sapere così evitiamo di perderci tempo in due.

brusarp, 2025-02-24:
Ciao!
ho provato con Grok a fargli leggere le richieste di EspHome e il Git di @lorenzo-deluca 
Ecco la risposta, non so se può tornare utile per prendere qualche spunto:

https://pastebin.com/DwPZUJsn

cannos88, 2025-02-26:
> Si, una volta capito come migrare con la nuova modalità stavo pensando di fare un altro file sorgente per la Kyo4 e Kyo8 base per semplificare un po il codice.
>
>
> Anche io ieri ho fatto qualche [...]

Riesci a girarmi queste modifiche cosi faccio un test?
Non ho fatto ancora nessun fork perchè ho lavorato direttamente su una copia di backup dei file del progetto.
Purtroppo però tutte le strade da me seguite non riuscivano a portare alla build.

diegoflyer, 2025-02-27:
Ho una KYO 32 ed ero pronto a installare tutto giusto un giorno dopo aver installato la 2025.2.0
Non sono un programmatore, se posso essere d'aiuto per testare qualcosa fatemi sapere...

lorenzo-deluca, 2025-02-27:
Nei prossimi giorni riprovo e magari chiedo anche agli sviluppatori di ESPHome più che altro perché vorrei evitare di dover far cambiare completamente il mapping dei sensori... comunque secondo me per ora per chi ancora non l'ha buildato conviene usare una versione precedente di ESPhome in locale sul PC tanto poi l'interfacciamento funziona anche con le ultime versioni, è solo un problema di build

adepiero, 2025-02-27:
> Ho una KYO 32 ed ero pronto a installare tutto giusto un giorno dopo aver installato la 2025.2.0 Non sono un programmatore, se posso essere d'aiuto per testare qualcosa fatemi sapere...

@diegoflyer, oltre alla soluzione proposta da @lorenzo-deluca, prova a dare un'occhiata a questo [ESPhome legacy](https://github.com/khenderick/esphome-legacy-addons?tab=readme-ov-file). Dovrebbe fare al caso tuo.

diegoflyer, 2025-02-27:
Grazie @adepiero e @lorenzo-deluca , valuto se iniziare così, non ho fretta perchè per ora la centrale è conessa via in/out hardware...

lcavalli, 2025-02-27:
Ci sarebbe anche questo https://github.com/robertklep/esphome-custom-component ma non l'ho provato personalmente.

gianlucaf81, 2025-02-28:
Piccolo OT

per chi usa ESPHome su container Proxmox si può fare il dowgrade alla versione 2024.12.4 
con i seguenti comandi dalla console del container ESPHome:

`pip uninstall -y esphome
`
`pip install esphome==2024.12.4`

controllare che la versione installata sia corretta con:

`esphome --version
`

infine riavviate il container.

lorenzo-deluca, 2025-03-02:
Ciao, ho pushato qui il workaround per far funzionare la build
https://github.com/lorenzo-deluca/espkyogate/tree/63-esphome-202520-e-custom-components
Ho usato il repository referenziato da @lcavalli, potrei anche portare tutto dentro questo, valuterò.
Usa ancora il custom_component ma dovrebbe buildare anche con le nuove versioni di ESPHome.

Migrare a external_component ci ho provato ma richiede parecchio lavoro e la configurazione va completamente rifatta.. sarebbe anche più comoda perché a quel punto aggiungi solo le aree e le zone ma è veramente parecchio lavoro, soprattuto ritestare tutto.. si accettano volontari! 🗡 

Intanto fatemi sapere se con questo workaround funziona a tutti che preparo la nuova versione.

Ciao!

closed 2025-03-02 by lorenzo-deluca

diegoflyer, 2025-03-02:
Ciao @lorenzo-deluca, ero riuscito anche io a far la build con il repository referenziato da @lcavalli ...
Oggi ho provato a installare (hardware) il tutto ma non so cosa è successo, perso codici e resettato la centrale a impostazioni di fabbrica, oggi son riuscito a ricaricare un backup, nei giorni prossimi riproverò...

brusarp, 2025-03-02:
> Ciao @lorenzo-deluca, ero riuscito anche io a far la build con il repository referenziato da @lcavalli ...
> Oggi ho provato a installare (hardware) il tutto ma non so cosa è successo, perso codici e [...]

Se per caso hai cortocircuitato i PIN 2 e 3 della seriale sulla centrale l'hai resettata completamente. È previsto come misura estrema per riportare la centrale alle condizioni di fabbrica 

[image]

lorenzo-deluca, 2025-03-03:
> Se per caso hai cortocircuitato i PIN 2 e 3 della seriale sulla centrale l'hai resettata completamente. È previsto come misura estrema per riportare la centrale alle condizioni di fabbrica
>
> [image]

Si, penso anche io che il problema possa essere stato causato dal corto di quei pin sul connettore seriale.
Però è molto strano che l'adattatore RS232 cortocircuiti quei pin.. vuol dire che è stato saldato male il connettore.
@diegoflyer magari prima di ricollegare l'adattore verifica con il multimetro la continuità.

Fatemi sapere, grazie, buona settimana :)

diegoflyer, 2025-03-03:
> Si, penso anche io che il problema possa essere stato causato dal corto di quei pin sul connettore seriale.
> Però è molto strano che l'adattatore RS232 cortocircuiti quei pin.. vuol dire che è stato [...]
@brusarp  @lorenzo-deluca 
Alcuni dettagli:
- l'unico errore che forse ho fatto è stato uscire dalla programmazione da tastiera con adattatore RS232 già inserito, ma credo non centri nulla
- il reset dovrebbe funzionare quando la centrale viene rialimentata, forse quando si esce dalla programmazione da tastiera viene comunque fatto un riavvio...
- ho provato con il tester i pin incriminati ma non mi risultano cortocircuitati, a meno che non succesa qualcosa al boot dell'ESP che per qualche istante vanno a massa
- l'adattatore che ho usato è questo: https://it.aliexpress.com/item/1005006513010110.html?spm=a2g0o.order_list.order_list_main.35.7fba3696iaNwHb&gatewayAdapt=glo2ita
  ne ho uno di scorta proverò con quello
- la scheda base che ho usato è un wt32 collegato sui pin 5 e 17 (ho anche un nodemcu esp8266, posso provare con quello):

[image]

Grazie dei vostri suggerimenti

adepiero, 2025-03-03:
> Ciao, ho pushato qui il workaround per far funzionare la build https://github.com/lorenzo-deluca/espkyogate/tree/63-esphome-202520-e-custom-components Ho usato il repository referenziato da [...]

Ciao @lorenzo-deluca. Pare funzionare tutto, almeno per me.
Attento però che c'è un errore. Le righe  da 9 a 11 corrette sono:

```
# ESP32 boards alternative directives
# esp32:
#   board: esp32doit-devkit-v1
```

lorenzo-deluca, 2025-03-03:
> Ciao [@lorenzo-deluca](https://github.com/lorenzo-deluca). Pare funzionare tutto, almeno per me. Attento però che c'è un errore. Le righe da 9 a 11 corrette sono:
>
> ```
> # ESP32 boards alternative [...]

Ciao @adepiero , quelle righe dipendono dalla scheda, ovviamente.
Il mio esempio era ancora basato sull'ESP8266, chiaramente se hai un'altra scheda va cambiato ;)

> > Si, penso anche io che il problema possa essere stato causato dal corto di quei pin sul connettore seriale.
> > Però è molto strano che l'adattatore RS232 cortocircuiti quei pin.. vuol dire che è stato saldato male il connettore.
> > [@diegoflyer](https://github.com/diegoflyer) magari prima di ricollegare l'adattore verifica con il multimetro la continuità.
> > Fatemi sapere, grazie, buona settimana :)
> 
> [@brusarp](https://github.com/brusarp) [@lorenzo-deluca](https://github.com/lorenzo-deluca) Alcuni dettagli:
> 
> * l'unico errore che forse ho fatto è stato uscire dalla programmazione da tastiera con adattatore RS232 già inserito, ma credo non centri nulla
> * il reset dovrebbe funzionare quando la centrale viene rialimentata, forse quando si esce dalla programmazione da tastiera viene comunque fatto un riavvio...
> * ho provato con il tester i pin incriminati ma non mi risultano cortocircuitati, a meno che non succeda qualcosa al boot dell'ESP che per qualche istante vanno a massa
> * l'adattatore che ho usato è questo: https://it.aliexpress.com/item/1005006513010110.html?spm=a2g0o.order_list.order_list_main.35.7fba3696iaNwHb&gatewayAdapt=glo2ita
>   ne ho uno di scorta proverò con quello
> * la scheda base che ho usato è un wt32 collegato sui pin 5 e 17 (ho anche un nodemcu esp8266, posso provare con quello):
> 
> [image]
> 
> Grazie dei vostri suggerimenti

Allora, sicuramente ogni volta che esci dalla programmazione la centrale si riavvia...
Ma non dovrebbe essere mai un problema, io ho la seriale collegata perennemente da anni e sono entrato e uscito dalla programmazione decine di volte, non si è mai resettata.

Non ho mai usato quell'ESP con interfaccia ethernet, in ogni caso, se vuoi un consiglio, non ricordo se l'avevo specificato in documentazione, NON usare i classici pin TX ed RX per collegarti al modulo seriale.. usa qualsiasi altro GPIO, tanto basta cambiare la sezione UART.
Questo perché in fase di startup il modulo manda su seriale trace di dump, non credo ci sia modo di bypassarlo, credo sia a livello di bootloader a basso livello.
Mentre se usi altri PIN dovresti riuscire a controllare meglio la gestione.

Fammi sapere come va... nel caso eventualmente sistemo la documentazione

diegoflyer, 2025-03-09:
@lorenzo-deluca sono riuscito a riprovare l'installazione ma la cosa strana è che ho provato ad entrare a uscire dalla programmazione solo con l'adattatore RS232 inserito (nè alimentato e nè collegato ai pin tx rx dell'ESP) e comunque ho il reset della centrale, ne avevo 2 adattatori e succede con entrambi, potrebbe essere l'adattatore fatto male visto che è leggermente diverso da quello che avevi postato nella documentazione? Qualcuno può indicarmi il modello preciso acquistato così provo? 
Questa la situazione che trovo dopo il reset:

[image]

[image]

La centrale:

[image]

lorenzo-deluca, 2025-03-09:
> [@lorenzo-deluca](https://github.com/lorenzo-deluca) sono riuscito a riprovare l'installazione ma la cosa strana è che ho provato ad entrare a uscire dalla programmazione solo con l'adattatore RS232 [...]

A questo punto penso sia molto probabile che sia l'adattatore.. le foto che hai postato direi che sono poco utili, piuttosto manda una foto dell'adattatore, no?

Comunque io ho diversi di [questi]( https://www.aliexpress.com/item/1005007666635845.html) e non ho mai avuto problemi di reset.

diegoflyer, 2025-03-09:
>  le foto che hai postato direi che sono poco utili, piuttosto manda una foto dell'adattatore, no?

L'adattatore è questo, lo avevo postato anche sopra ;):
https://it.aliexpress.com/item/1005006513010110.html?spm=a2g0o.order_list.order_list_main.41.462b3696SOd3pB&gatewayAdapt=glo2ita

lorenzo-deluca, 2025-03-09:
> > le foto che hai postato direi che sono poco utili, piuttosto manda una foto dell'adattatore, no?
> 
> L'adattatore è questo, lo avevo postato anche sopra ;): https://it.aliexpress.com/item/1005006513010110.html?spm=a2g0o.order_list.order_list_main.41.462b3696SOd3pB&gatewayAdapt=glo2ita
> 
> grazie, provo ad acquistare l'item che usi tu

Si, intendevo anche magari come l'hai collegato all'esp.. comunque quello mi sembra identico al mio.
Come l'hai alimentato? Se non ricordo male è meglio alimentarlo sul 5V, era già stato aperto qualche topic in merito.
Inoltre, vedo che hai un cavo collegato alla DB9 della centrale, se è un cavo per PC probabilmente è null-modem.. come da schema l'adattatore seriale va collegato DIRETTAMENTE alla porta DB9 sulla centrale, senza cavi di mezzo, o meglio, se metti un cavo questo deve essere dritto.

Secondo me è un problema di collegamento non tanto di adattatore..

edmcska, 2025-03-11:
> Ciao, ho pushato qui il workaround per far funzionare la build https://github.com/lorenzo-deluca/espkyogate/tree/63-esphome-202520-e-custom-components Ho usato il repository referenziato da [...]

Hello Lorenzo,

Your workaround looks to be working very well. Still in test phase :d ,but so far so good.

Thanks , and greetings from Hungary,

Adam

ddtterry, 2025-03-11:
Ciao. Mi sto approcciando, solo grazie al progetto kyo 32di Lorenzo, al mondo homeassistat e esp32. Volevo chiedervi se nel frattempo wualcuno ha avuto modo di sperimentare su kyo320 e se , in generale,  esiste qualche guida in italiano sempre per centrali bentel. Grazie.

jovigiso, 2025-03-24:
Hello, my Kyo configuration it was working ok, my SD is dead and no backup,so I need to configure everything. Home Assistant 2025.03.04 and I've done the upgrade with GitHub new files, I have 10 zones but just the first 4 work. Any ideias? Thanks

lorenzo-deluca, 2025-03-27:
> Hello, my Kyo configuration it was working ok, my SD is dead and no backup,so I need to configure everything. Home Assistant 2025.03.04 and I've done the upgrade with GitHub new files, I have 10 zones but just the first 4 work. Any ideias? Thanks

please open a dedicated issue and post your configuration file. thanks

jovigiso, 2025-03-27:
Hello, I solve it!

The original yaml only have &kyo32->zona[0] to zona[3], and I add  until zona [12], and is working like before.

Thank you very much for your help.

Regards,
José Gonçalves