lorenzo-deluca/espkyogate #64: alarm control panel

ciano68, 2025-02-22:
salve lorenzo sto usando espkyogate su   kyo8gw , vedo i sensori  e funzionano bene , io però vorrei usare il panello di controllo (alarm control panel) per poter armare , disarmare e controllare lo stato anche se inserisco l'allarme con la chiave fisica"SAT" se qualcuno può aiutarmi.

lorenzo-deluca, 2025-02-24:
Ciao, onestamente non so come funzioni la chiave SAT... mai avuta.
Prima di chiedere a GPT dovresti però capire come fare per ottenere quello che ti serve manualmente: chiamata servizi, stati, etc..
Poi una volta capito fare le automazioni viene di conseguenza, non so se GPT conosce le logiche Bentel o del mio firmware..

Magari qualcuno qui che ha la chiave SAT sa darti supporto.

ciano68, 2025-02-24:
Io intendevo con attivazione antifurto 
tramite inseritore Bentel o da tastiera

ciano68, 2025-02-24:
Quello che io vorrei avere è conoscere quando l'antifurto è attivato e la possibilità di attivare/ disattivare con la tastiera e codice su home Assistant

lorenzo-deluca, 2025-02-24:
Ho capito quello che vuoi fare, ma per farlo devi prima capire come funziona l'integrazione e configurarla di conseguenza.

Se hai configurato come da documentazione i sensori delle aree che ti interessano dovresti avere tutte le informazioni che ti servono per ricostruire lo stato di attivazione delle varie aree.

Ripeto, essendo una cosa piuttosto specifica non so quanto ti possa dare la soluzione GPT... ti conviene riguardare la documentazione e vedere se gli stati dei sensori rispecchiano lo stato della tua centrale.

lorenzo-deluca, 2025-03-22:
issue chiusa.

closed 2025-03-22 by lorenzo-deluca