lorenzo-deluca/espkyogate #65: Adattatore su KYO32 genera il reset della centrale

diegoflyer, 2025-03-09:
Faccio Riassunto da issue precedente dove questo tema ero off-topic:

Uscendo dalla programmazione da tastiera con adattatore RS232 la centrale resetta ai dati di fabbrica.
Questo potrebbe essere generato da un adattatore difetttoso in quanto un eventuale cortocircuito di alcuni pin di fatto genera reset della centrale:

[image]

ho provato con il tester i pin incriminati ma non mi risultano cortocircuitati, a meno che non succeda qualcosa al boot dell'ESP che per qualche istante vanno a massa
l'adattatore che ho usato è questo:
https://it.aliexpress.com/item/1005006513010110.html?spm=a2g0o.order_list.order_list_main.35.7fba3696iaNwHb&gatewayAdapt=glo2ita
ho provato anche con un altro adattatore identico e succede la stessa cos
la scheda base che ho usato è un wt32 collegato sui pin 5 e 17 (ho anche un nodemcu esp8266, posso provare con quello):

[image]

Grazie dei vostri suggerimenti

diegoflyer, 2025-03-09:
Questa l'immagine della centrale (in questo foto c'è inserito il cavo di programmazione da PC mentre la stavo riprogrammando dopo il reset):

[image]

> Si, intendevo anche magari come l'hai collegato all'esp.. comunque quello mi sembra identico al mio.
> Come l'hai alimentato? Se non ricordo male è meglio alimentarlo sul 5V, era già stato aperto qualche topic in merito.
> Inoltre, vedo che hai un cavo collegato alla DB9 della centrale, se è un cavo per PC probabilmente è null-modem.. come da schema l'adattatore seriale va collegato DIRETTAMENTE alla porta DB9 sulla centrale, senza cavi di mezzo, o meglio, se metti un cavo questo deve essere dritto.
> Secondo me è un problema di collegamento non tanto di adattatore.
> 
> Comunque, è off-topic rispetto a questa issue.. se vuoi aprire un'altra issue così che possa essere utile anche ad altri utenti, comunque qualcosa se ne era parlato qui https://github.com/lorenzo-deluca/espkyogate/issues/30

il problema che mi fa pensare che è l'adattatore il problema è che il reset avviene con l'adattatore solo inserito ma con nessuno dei 4 pin collegato (quindi nè alimentato nè collegato al tx rx dell'ESP).
Potrebbe essere che se non alimentato si comporta in questo modo?

lorenzo-deluca, 2025-03-22:
@diegoflyer  hai aggiornamenti?

closed 2025-03-22 by lorenzo-deluca