lorenzo-deluca/espkyogate #67: Esperienza con Bentel Kyo 320

ddtterry, 2025-03-24:
Ciao. Mi sto approcciando, solo grazie al progetto kyo 32 di Lorenzo, al mondo homeassistat e esp32. Volevo chiedervi se nel frattempo qualcuno ha avuto modo di sperimentare su kyo320 e se , in generale,  esiste qualche guida aggiornata in italiano sempre per centrali bentel. 
Dovrei avere una 32G da qualche parte per sperimentare per poi integrare la 320.
Grazie.

lorenzo-deluca, 2025-03-25:
Ciao, mi sembra un'ottima idea quella di partire dalla Kyo32 per sperimentare, sulla Kyo32 sicuramente funziona tutto.
Per quanto riguarda la Kyo320 non ho feedback, fammi sapere come va! :)

ddtterry, 2025-04-09:
Ciao.  Sono riuscito a collegare senza grossi problemi la kyo32 ed ho iniziato a modificare il file yaml in base a diverse esigenze. Purtroppo quando ho provato a collegare una kyo 320 nel log mi dice nessuna risposta.
Forse i pins della seriale non sono gli stessi? O forse serve inserire la password utente?
Hai qualche suggerimento da darmi. Grazie

lorenzo-deluca, 2025-04-09:
> Ciao. Sono riuscito a collegare senza grossi problemi la kyo32 ed ho iniziato a modificare il file yaml in base a diverse esigenze. Purtroppo quando ho provato a collegare una kyo 320 nel log mi [...]

Non avendo mai visto la Kyo 320 non ti so dire molto, i PIN della RS232 dovrebbero essere sempre quelli.. sicuramente è più un problema di protocollo, hai provato a abilitare i log seriali?

``yaml
action: esphome.espkyogate_debug_command
data:
  serial_trace: 1
  log_trace: 1
  polling_kyo: 1
``

ddtterry, 2025-04-11:
No stasera provo e poi ti dico grazie

ddtterry, 2025-04-11:
Ciao 
Ho aggiunto le righe ma facendo validate mi dice questo:

INFO ESPHome 2025.5.0-dev
INFO Reading configuration /config/esphome/espkyogate.yaml...
INFO Unable to import component action: No module named 'esphome.components.action'
Failed config

action: [source /config/esphome/espkyogate.yaml:50]
  
  Component not found: action.
  esphome.espkyogate_debug_command

lorenzo-deluca, 2025-04-14:
Ciao @ddtterry, ho dimenticato di specificare che il nome del servizio cambia il base a come hai chiamato il dispositivo.. se scrivi "esphome" nei servizi lo trovi.. è quello che finisce con debug_command

ddtterry, 2025-04-14:
Ciao Lorenzo.
Non riesco a trovare il servizio.
E' sotto dispositivi=> servizi?
Scusa ma sono neofiti di ha.
Grazie

ddtterry, 2025-04-14:
ok trovato grazie ma è come il tuo.
esphome.espkyogate_debug_command
Ma il risultato non cambia

INFO ESPHome 2025.3.3
INFO Reading configuration /config/esphome/espkyogate.yaml...
INFO Unable to import component action: No module named 'esphome.components.action'
Failed config

action: [source /config/esphome/espkyogate.yaml:57]
  
  Component not found: action.
  esphome.espkyogate_debug_command

ddtterry, 2025-04-14:
ho provato a compilarlo direttamente dallo strumento sviluppatore ma quando faccio "esegui azione" mi da unknow error

ddtterry, 2025-04-14:
ok ci sono riuscito mi da il verde. il problema era che non avevo la kyo 32 e la esp32 alimentati.
Solo che passando su 320 mi dice sempre no answer su serial port. che nervi.
Altri suggerimenti?

ddtterry, 2025-04-14:
probabilmente esphome non accetta più l'abilitazione dei log direttamente nello yaml

lorenzo-deluca, 2025-04-14:
> probabilmente esphome non accetta più l'abilitazione dei log direttamente nello yaml

no, dovrebbe.. a me funziona.. comunque direi che la 320 non funziona e morta lì.. se hai una Kyo32 tanto meglio 👍

closed 2025-04-14 by lorenzo-deluca

ddtterry, 2025-04-14:
Magari. ho una 320 che non voglio assolutamente cambiare tant'è che ne ho comprata un'altra da tenere come scorta visto che è uscita di produzione.
Nessuna absoluta, secondo me, è ancora a livello della 320. Forse la nuova che ancora deve uscire.
Forse essendo equipaggiata con scheda di rete è meglio puntare con integrazioni di protocollo tipo SIA ma, se non erro, SIA lavora solo in ascolto.
Grazie comunque di tutto.
Ciao

lorenzo-deluca, 2025-04-16:
Ti dico, non è che mi intenda troppo di centrali di allarme, mi sono trovato questa perchè era la più economica ed era parzialmente già installata in casa.
Onestamente mi trovo bene con la Kyo32, ovviamente con la mia integrazione, da quel che ho visto con HA hai più funzionalità delle nuove che costano migliaia di €, anche io ho comprato varie schede a due lire usate e sono a posto per anni.

Detto ciò, se la 320 ha già un'interfaccia di rete sicuramente si può integrare tramite protocolli che non siano quello seriale, ho visto ceh c'è sia l'[integrazione nativa HA]( https://www.home-assistant.io/integrations/sia/) che [una custom](https://github.com/Cheaterdev/sia-ha)

Ora, dato che non avendola di più non posso fare chiudo la issue :)