lorenzo-deluca/espkyogate #74: Available services

serucci, 2025-07-14:
Ciao, mi trovo a dover escludere alcune zone durante alcuni tipi di parzializzazioni ,ma su home assistant non riesco a puntare al service zone_exclude/include in quanto , differenza di tutti gli altri, non mi compare.
Qualcuno ha già avuto modo di testare questo service ?

serucci, 2025-07-19:
No ,non mi appare vedo tutti i service ma non quello 

[image]

[image]

non capisco

lorenzo-deluca, 2025-07-19:
> No ,non mi appare vedo tutti i service ma non quello
> 
> [image]
> 
> [image]
> 
> non capisco

A me compaiono su entrambe le centrali:

[image]

serucci, 2025-08-22:
Nel file bentel_32.h  che uso  la "chiamata" del servizio è disabilitata

[image]

se l'abilito e compilo mi da questo errore 

[image]