lorenzo-deluca/espkyogate #76: ESPHome Device Builder 2025.7.2 fails to compile

lorisdrg, 2025-07-19:
I dowloaded the repo at commit [eb54990](https://github.com/lorenzo-deluca/espkyogate/commit/eb54990a02e2a7ed3ed6c1e7be11f51fe6ad1e5d), the latest at this time, and pasted in the esphome folder under the config of my home assistant instance (Installation method: Home Assistant OS, Core: 2025.7.2, Supervisor: 2025.07.1, Operating System: 16.0, Frontend: 20250702.2, HW: raspberry pi 5). Then I just added the secrets.yaml file and corrected the espkyogate_configuration.yaml file with the uart pin for the esp32. I got the following error

```
INFO ESPHome 2025.7.2
INFO Reading configuration /config/esphome/espkyogate_configuration.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing espkyogate (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.10.2 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.10
Linking .pioenvs/espkyogate/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o:(.literal._ZN7esphome3api15UserServiceBaseIJiiEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJiiEE28encode_list_service_responseEv]+0x0): undefined reference to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o:(.literal._ZN7esphome3api15UserServiceBaseIJiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE[_ZN7esphome3api15UserServiceBaseIJiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE]+0x0): undefined reference to `_ZN7esphome3api21get_execute_arg_valueIiEET_RKNS0_22ExecuteServiceArgumentE'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o: in function `_ZN7esphome10EntityBase23set_disabled_by_defaultEb':
/data/build/espkyogate/src/esphome/core/entity_base.h:46:(.text._ZN7esphome3api15UserServiceBaseIJiiEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJiiEE28encode_list_service_responseEv]+0x16): undefined reference to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /data/build/espkyogate/src/esphome/core/entity_base.h:46:(.text._ZN7esphome3api15UserServiceBaseIJiiEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJiiEE28encode_list_service_responseEv]+0x1e): undefined reference to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o: in function `_ZN7esphome11Application18register_componentINS_6uptime19UptimeSecondsSensorEEEPT_S5_':
/data/build/espkyogate/src/esphome/core/application.h:292:(.text._ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv]+0x17): undefined reference to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /data/build/espkyogate/src/esphome/core/application.h:292:(.text._ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv]+0x1f): undefined reference to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o: in function `_ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv':
/config/esphome/espkyogate_configuration.yaml:552:(.text._ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv]+0x27): undefined reference to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o:/config/esphome/espkyogate_configuration.yaml:556: more undefined references to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv' follow
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o: in function `_ZN7esphome3api15UserServiceBaseIJiiEE8execute_IJLi0ELi1EEEEvSt6vectorINS0_22ExecuteServiceArgumentESaIS5_EENS_3seqIJXspT_EEEE':
/data/build/espkyogate/src/esphome/components/api/user_services.h:58:(.text._ZN7esphome3api15UserServiceBaseIJiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE[_ZN7esphome3api15UserServiceBaseIJiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE]+0x2b): undefined reference to `_ZN7esphome3api21get_execute_arg_valueIiEET_RKNS0_22ExecuteServiceArgumentE'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /data/build/espkyogate/src/esphome/components/api/user_services.h:58:(.text._ZN7esphome3api15UserServiceBaseIJiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE[_ZN7esphome3api15UserServiceBaseIJiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE]+0x36): undefined reference to `_ZN7esphome3api21get_execute_arg_valueIiEET_RKNS0_22ExecuteServiceArgumentE'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o: in function `_ZN7esphome3api15UserServiceBaseIJiiiEE8execute_IJLi0ELi1ELi2EEEEvSt6vectorINS0_22ExecuteServiceArgumentESaIS5_EENS_3seqIJXspT_EEEE':
/data/build/espkyogate/src/esphome/components/api/user_services.h:58:(.text._ZN7esphome3api15UserServiceBaseIJiiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE[_ZN7esphome3api15UserServiceBaseIJiiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE]+0x2a): undefined reference to `_ZN7esphome3api21get_execute_arg_valueIiEET_RKNS0_22ExecuteServiceArgumentE'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /data/build/espkyogate/src/esphome/components/api/user_services.h:58:(.text._ZN7esphome3api15UserServiceBaseIJiiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE[_ZN7esphome3api15UserServiceBaseIJiiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE]+0x35): undefined reference to `_ZN7esphome3api21get_execute_arg_valueIiEET_RKNS0_22ExecuteServiceArgumentE'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /data/build/espkyogate/src/esphome/components/api/user_services.h:58:(.text._ZN7esphome3api15UserServiceBaseIJiiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE[_ZN7esphome3api15UserServiceBaseIJiiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE]+0x42): undefined reference to `_ZN7esphome3api21get_execute_arg_valueIiEET_RKNS0_22ExecuteServiceArgumentE'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o:/data/build/espkyogate/src/esphome/components/api/user_services.h:58: more undefined references to `_ZN7esphome3api21get_execute_arg_valueIiEET_RKNS0_22ExecuteServiceArgumentE' follow
collect2: error: ld returned 1 exit status
*** [.pioenvs/espkyogate/firmware.elf] Error 1
========================= [FAILED] Took 17.80 seconds =========================

```

lorenzo-deluca, 2025-07-19:
Try to remove

platformio_options:
    board_build.f_cpu: 160000000L

Let me know ;)

lorisdrg, 2025-07-19:
I removed that part but still failing:
```
INFO ESPHome 2025.7.2
INFO Reading configuration /config/esphome/espkyogate_configuration.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing espkyogate (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.10.2 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.10
Linking .pioenvs/espkyogate/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o:(.literal._ZN7esphome3api15UserServiceBaseIJiiEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJiiEE28encode_list_service_responseEv]+0x0): undefined reference to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o:(.literal._ZN7esphome3api15UserServiceBaseIJiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE[_ZN7esphome3api15UserServiceBaseIJiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE]+0x0): undefined reference to `_ZN7esphome3api21get_execute_arg_valueIiEET_RKNS0_22ExecuteServiceArgumentE'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o: in function `_ZN7esphome10EntityBase23set_disabled_by_defaultEb':
/data/build/espkyogate/src/esphome/core/entity_base.h:46:(.text._ZN7esphome3api15UserServiceBaseIJiiEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJiiEE28encode_list_service_responseEv]+0x16): undefined reference to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /data/build/espkyogate/src/esphome/core/entity_base.h:46:(.text._ZN7esphome3api15UserServiceBaseIJiiEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJiiEE28encode_list_service_responseEv]+0x1e): undefined reference to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o: in function `_ZN7esphome11Application18register_componentINS_6uptime19UptimeSecondsSensorEEEPT_S5_':
/data/build/espkyogate/src/esphome/core/application.h:292:(.text._ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv]+0x17): undefined reference to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /data/build/espkyogate/src/esphome/core/application.h:292:(.text._ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv]+0x1f): undefined reference to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o: in function `_ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv':
/config/esphome/espkyogate_configuration.yaml:550:(.text._ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJiiiEE28encode_list_service_responseEv]+0x27): undefined reference to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o:/config/esphome/espkyogate_configuration.yaml:554: more undefined references to `_ZN7esphome3api19to_service_arg_typeIiEENS0_5enums14ServiceArgTypeEv' follow
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o: in function `_ZN7esphome3api15UserServiceBaseIJiiEE8execute_IJLi0ELi1EEEEvSt6vectorINS0_22ExecuteServiceArgumentESaIS5_EENS_3seqIJXspT_EEEE':
/data/build/espkyogate/src/esphome/components/api/user_services.h:58:(.text._ZN7esphome3api15UserServiceBaseIJiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE[_ZN7esphome3api15UserServiceBaseIJiiEE15execute_serviceERKNS0_21ExecuteServiceRequestE]+0x2b): undefined reference to `_ZN7esphome3api21get_execute_arg_valueIiEET_RKNS0_22ExecuteServiceArgumentE'
[... cut ...]
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/espkyogate/src/main.cpp.o:/data/build/espkyogate/src/esphome/components/api/user_services.h:58: more undefined references to `_ZN7esphome3api21get_execute_arg_valueIiEET_RKNS0_22ExecuteServiceArgumentE' follow
collect2: error: ld returned 1 exit status
*** [.pioenvs/espkyogate/firmware.elf] Error 1
========================= [FAILED] Took 17.27 seconds =========================

```

lorisdrg, 2025-07-20:
I was able to build by changing in the _bentel_kyo32.h_ the type of variable form _int_ to _float_ for the following methods:
```
arm_area
disarm_area
activate_output
deactivate_output
pulse_output
debug_command
update_datetime
include_zone
exclude_zone
```
I still have to apply hardware changes to my Bentel alarm system to test this build, then I let you know. In the meantime, do you think that could be a valid solution the change of variable type or could it have side effects?

closed 2025-07-20 by lorenzo-deluca