lorenzo-deluca/espkyogate #79: Register Service not declared

pow3r77, 2025-07-23:
Good morning,
i have this issue in the last update:

In file included from src/main.cpp:99:
src/bentel_kyo32.h: In member function 'virtual void Bentel_Kyo32::setup()':
src/bentel_kyo32.h:66:4: **error: 'register_service' was not declared in this scope**
   66 |    register_service(&Bentel_Kyo32::arm_area, "arm_area", {"area", "arm_type", "specific_area"});
      |    ^~~~~~~~~~~~~~~~
*** [.pioenvs/espkyogatebox/src/main.cpp.o] Error 1
========================== [FAILED] Took 8.21 seconds ==========================
======== [ERROR] /config/esphome/espkyogatebox.yaml ========

lorenzo-deluca, 2025-07-23:
Hello @pow3r77, look at YAML example configuration, already discussed, and resolved, in #75 👍

closed 2025-07-23 by lorenzo-deluca

pow3r77, 2025-07-23:
Great! thank you!